espressif/esp-idf #3585: [NIMBLE] build warnings and extra debug logging

no1seman, 2019-06-04:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):  v3.3-beta1-462-g127caf3a0 + latest nimble branch: 127caf3a04c603b8c289a67d36849c3ef9f269d7
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB|external 5V&Battery

## Problem Description
Continuing to explore NIMBLE port and got the following issues:
1. Can't disable debug logging produced by BLE_HS_LOG macro from ble_hs_log.h (except changing component source code):
....
GAP procedure initiated: stop advertising.
Device Address: b4:e6:2d:ee:7e:77
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
...
2. Got multiple redefinition build warnings if try to add any network components to project that uses nimble:

[build.zip](https://github.com/espressif/esp-idf/files/3253648/build.zip)

rom/queue.h overrides nimble/include/os/queue.h definitions
and
lwip/lwip/src/include/lwip/def.h overrides nimble/include/os/endian.h definitions

3. Vanilla nimble 1.1.0 have support for extended packets (with 251 byte length, BLE 5.0) advertisement and discovery and also set of functions to setup periodic advertisment (http://mynewt.apache.org/latest/network/docs/ble_hs/ble_gap.html). In esp32 port they are unavaible.

### Expected Behavior
1. A way to disable debug logs
2. No any build warnings
3. All vanilla Nimble GAP functions have to be avaible 
### Actual Behavior
1. Debug logs can't be disabled
2. Multiple build warnings
3. BLE 5.0 specific functions is unavaible
### Steps to repropduce

1. just run any nimble example
2. Simply add #include "tcpip_adapter.h" in any nimble example
3. MYNEWT_VAL(BLE_EXT_ADV) in ble_gap.h is false so functions is not avaible

### Code to reproduce this issue
## Debug Logs
## Other items if possible

dhrishi, 2019-06-13:
@no1seman Sorry for the delayed response

1. Did you try to disable logging from menuconfig? That should work:
make menuconfig -> Component config -> Log output -> Default log verbosity -> No output

2. Thanks. We will fix this.

3. ESP32 controller does not support extended advertising. Even though we enable this in host, it won't work and that's why it is kept disabled.

no1seman, 2019-06-13:
@dhrishi,
1. Yes you are right setting Default log verbosity to No output turn off these logs but it also turns off ALL logs. All esp-idf components have an ability to set logging level via esp_log_level_set() and log messages produced with BLE_HS_LOG macro can't be tune that way. It's a little uncomfortable.
2. rom/queue.h overrides nimble/include/os/queue.h definitions - fixed by last commits
and
lwip/lwip/src/include/lwip/def.h overrides nimble/include/os/endian.h definitions - ok, will wait
3. If there any plans to support BLE 5.0?

alyf80, 2020-11-24:
FYI, NimBLE logging is still very much a problem with the version in the current 4.1 head (c3324a82a3e6beb30d5c24b9828c01ced3864d7f)

>     1. Did you try to disable logging from menuconfig? That should work:
>        make menuconfig -> Component config -> Log output -> Default log verbosity -> No output

Well, yes, but that's hardly a solution. This problem only happens if CONFIG_LOG_DEFAULT_LEVEL was changed from its default level of INFO to something higher, such as DEBUG, which is generally done for a reason (in my case, I need to collect debug-level logs from another IDF component).

NimBLE logging level should really be controllable either at runtime through esp_log_level_set() or at compile time via a separate config option.

prasad-alatkar, 2020-11-24:
Hi @alyf80 you can try using `esp_log_level_set("NimBLE", ESP_LOG_NONE)` to disable NimBLE logging. Please let me know if it helps. Having said that, yes custom logging for NimBLE would be a great idea, will update you once we add this feature.

alyf80, 2020-11-25:
@prasad-alatkar digging around a bit more it turns out that custom logging for NimBLE has already been implemented, but then we have this in 
`components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c`, line 45, function  `void nimble_init_port(void)`:
```
#ifdef ESP_PLATFORM
    esp_log_level_set("NimBLE", LOG_LOCAL_LEVEL);
#endif
```
So NimBLE is forcing its log level back to the default whenever it gets initialized -- which in my case is well after log levels are configured in `app_main()`.

Now, I can change the log level back after calling `nimble_port_init()`, but I really think that code should not be there to begin with.

ankit-thealchemist, 2021-04-10:
> @prasad-alatkar digging around a bit more it turns out that custom logging for NimBLE has already been implemented, but then we have this in [...]

Yes indeed. There should be a way to control the log level of the nimble in the menuconfig itself . Just like the bluedroid did. I was working on the ble mesh, this create so much of the headache in debugging.

prasad-alatkar, 2021-04-14:
@ankit-thealchemist We are working on custom logging for NimBLE component, I will update you as soon as the changes are merged into master branch.

rahult-github, 2022-05-23:
Hi @ankit-thealchemist ,

Multiple fixes have gone in to improve the logging for NimBLE component. Please check this issue on latest releases to confirm. Closing this ticket now, please feel free to reopen if any issue still found. 

Thanks,
 Rahul

closed 2022-05-23 by rahult-github