espressif/esp-idf #3586: “pm_on_beacon_rx” appears during UART communication, abort() program

GitOfLance, 2019-06-05:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
I use the WiFi function of esp32 to receive data from `mqtt broker` and send them out after data processing. 
Because of the limitation of the target machine, only **`28Byte`** can be sent at a time. 
The total amount of data is **`50KB`**. These data will be sent out continuously. 
The baud rate is `115200*2`. In the process of sending, the following errors will occur accidentally.
[image]
As shown in the picture above:
I'm going to send `49178` Byte data. `abort()` appears when I send it to `8947`**(>>>number = 8947)**.
This problem appears irregularly. Sometimes it's okay to test for a whole day. Sometimes it comes up again and again.
I set up WiFi with default configuration，like this:
```
tcpip_adapter_init();
ret = esp_event_loop_init(wifi_event_handler, NULL);
if (ret != ESP_OK)
    goto WIFI_ERR_INIT;

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ret = esp_wifi_init(&cfg);
if (ret != ESP_OK)
    goto WIFI_ERR_INIT;

ret = esp_wifi_set_mode(WIFI_MODE_STA);
if (ret != ESP_OK)
    goto WIFI_ERR_INIT;

ret = esp_wifi_set_storage(WIFI_STORAGE_FLASH); 
if (ret != ESP_OK)
    goto WIFI_ERR_INIT;
ret = esp_wifi_start();

```
I configure the network through smart config, and when the WiFi connection succeeds, I get data from mqtt broker and send it via uart...
What does this error message mean `(pm_on_beacon_rx)`? Why did it happen? How to avoid it?

liuzfesp, 2019-06-06:
Hi @GitOfLance this issue has been fixed in 5d43ec64, you can update it to latest master to check it.

GitOfLance, 2019-06-06:
Hi @liuzfesp 
Ok, thank you very much, I have pulled the latest master, and I will give feedback here after the test.

Alvin1Zhang, 2019-06-13:
@GitOfLance Thanks for reporting the issue, could you help share if you already have the testing results? Thanks.

GitOfLance, 2019-06-18:
@Alvin1Zhang 
Sorry, I am late, this flag will still appear in uart communication, although it will not cause uart communication failure, but will restart the wifi module, but I did not save the log in time.

GitOfLance, 2019-06-18:
In addition, I allow to initialize wifi repeatedly, some of the code is as follows:
```
    /* 先终止wifi,适应重复初始化 */
    if (gse_reconnect & eventbits)
    {
            printf("wifi connected, stop and connect");
	    esp_wifi_stop();
    }

   /* 初始化wifi */
    tcpip_adapter_init();
    if (1 == first)
    {
	    ret = esp_event_loop_init(wifi_event_handler, NULL);
	    //first = 0;
     }
    printf("initialization number : [[%d]]\n", first);
    first++;
    if  (ret != ESP_OK)
	goto WIFI_ERR_INIT;

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ret = esp_wifi_init(&cfg);
    if  (ret != ESP_OK)
		goto WIFI_ERR_INIT;

    ret = esp_wifi_set_mode(WIFI_MODE_STA);
    if  (ret != ESP_OK)
		goto WIFI_ERR_INIT;
    ret = esp_wifi_set_storage(WIFI_STORAGE_FLASH);	
    if  (ret != ESP_OK)
		goto WIFI_ERR_INIT;
    ret = esp_wifi_start();	
```

I guarantee that the function `esp_event_loop_init` is called only once, and the `esp_wifi_stop` function is called to terminate the wifi module before repeating the initialization, and then repeat the wifi initialization operation.
In the previous version, there was no problem with this process, but there was a problem with this method after updating to the latest code:
[image]

liuzfesp, 2019-06-18:
@GitOfLance， maybe you need to call esp_wifi_deinit() after esp_wifi_stop().

GitOfLance, 2019-06-18:
@liuzfesp 
After adding the function `esp_wifi_deinit`, this prompt still exists, not sure whether it affects normal functions.
[image]
In addition, this problem sometimes occurs when using `smartconfig`:
[image]
This problem often occurs, and configuring wifi via smartconfig often fails, even when there is no such error.

liuzfesp, 2019-07-10:
HI @GitOfLance, could you let me know your IDF commit ID?

GitOfLance, 2019-07-12:
Hi, @liuzfesp 
[image]

liuzfesp, 2019-07-14:
Hi @GitOfLance we have found the root cause of this issue, hopefully the fix will be merged into IDF master in 1 week.

liuzfesp, 2019-07-25:
HI @GitOfLance, this issue has been fixed in 034440b8

Alvin1Zhang, 2019-07-25:
@GitOfLance Thanks for reporting the issue, feel free to reopen the ticket if your issue still exists. Thanks.

closed 2019-07-25 by Alvin1Zhang