espressif/esp-idf #3587: tiT blocking CPU1 and error stopping wifi

piltom, 2019-06-05:
# Environment

- Development Kit:      ESP32-DevKitC
- Kit version v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-667-gda13efc17
- Build System:      Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:    Linux
- Power Supply:         USB

## Problem Description

When stopping soft ap, the following error appears on monitor:
`E (30352) tcpip_adapter: handle_ap_stop 134 esp_wifi_internal_reg_rxcb ret=0x3014`

Later, when trying to connect in sta mode to another ap, the watchdog gets triggered because tiT (tcp/ip task) is hanging the cpu1:
```
E (42482) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

E (42482) task_wdt:  - IDLE1 (CPU 1)

E (42482) task_wdt: Tasks currently running:

E (42482) task_wdt: CPU 0: IDLE0

E (42482) task_wdt: CPU 1: tiT
```

### Expected Behavior
Program description:
The program implements a state machine that loads the last pair of ssid/pass from nvs, if it can connect it sends a request and then turns off. If it can't connect, it stops the station mode, starts a soft ap with an http server for setting the ssid/pass. When that value pair is posted, it saves them to nvs, turns off soft ap, and goes back to the initial state (except nvs load, which is called only once on startup).

For connecting to an ap in sta mode, i implemented a component, which works on its own (tested). It receives ssid, pass, max number of tries and a callback pointer. It tries to connect an amount of times and if it fails, it calls the callback and then calls esp_wifi_stop().

For creating the http server, I implemented 2 components, one for the server itself and one for wifi in ap mode. When a combination of new ssid and pass is entered, the server shuts itself down and again calls esp_wifi_stop() (and a callback to notify the state machine of the new config)
### Actual Behavior
-> Try to connect to the saved ssid and pass, failed (as expected)
-> start soft ap and set up web server.
-> After entering the configuration, post request is received and data is processed

### Steps to repropduce

### Code to reproduce this issue
I would have to ask for permission for this. I've logged a lot of events in the debug logs, it contains a lot of information.
Every time the handler for the connection is called, I log the event and event id.
The component for connecting to the wifi uses a modified version of the simple connect example.

## Debug Logs
What's going on on the log:
Trying 2 times to connect to the saved AP. Logged a WIFI_EVENT, with id 3 (sta stop)
After failing this, it creates the server, i connect and enter the values, pass them through a post request, and the values for ssid/pass are right (checked through logging).
Here starts the failing:
I call again for esp_wifi_stop(). Two WIFI_EVENT are logged, 3 and 14 (sta stop and ap sta dissconected).
Then i get this E log:
tcpip_adapter: handle_ap_stop 134 esp_wifi_internal_reg_rxcb ret=0x3014

Then another WIFI_EVENT with id 12 (ap stop).
Then the same E log as before.
After this, the code for connecting to the ap starts again (which works if you give it a valid ssid/pass the first time).

```
I (28) boot: ESP-IDF v4.0-dev-667-gda13efc17 2nd stage bootloader
I (28) boot: compile time 16:10:31
I (28) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
[... 108 lines cut ...]
I (37821) wifi station: ESP_WIFI_MODE_STA
W (37821) event: handler already registered, overwriting
W (37821) event: handler already registered, overwriting
W (37841) event: handler already registered, overwriting
I (37891) wifi: mode : sta (30:ae:a4:d3:3a:b4)
I (37891) wifi station: eventbase WIFI_EVENT, event id 2
I (37891) wifi station: wifi_init_sta finished.
I (37891) wifi station: eventbase WIFI_EVENT, event id 2
I (37901) wifi station: connect to ap SSID:maskaw password:dOmNJDfu
I (38501) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (39491) wifi: state: init -> auth (b0)
I (39501) wifi: state: auth -> assoc (0)
I (39511) wifi: state: assoc -> run (10)
I (43531) wifi: state: run -> init (2a0)
I (43531) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (43531) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
E (49501) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (49501) task_wdt:  - IDLE1 (CPU 1)
E (49501) task_wdt: Tasks currently running:
E (49501) task_wdt: CPU 0: IDLE0
E (49501) task_wdt: CPU 1: tiT

```

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3256086/sdkconfig.zip)

piltom, 2019-06-05:
I'm setting up a JTAG debugger at the moment, will update.

piltom, 2019-06-05:
Suspending the program during the watchdog trigger suggests that the tcp/ip task might be stuck in [igmp_tmr](http://www.nongnu.org/lwip/2_0_x/igmp_8c.html#a4418a22d37098f05d0c2fcfe288d0ca1).

The error could be related to [this](http://lwip.100.n7.nabble.com/TCP-tmr-endless-loop-td4091.html).

I will continue to debug, however i feel like this kind of error should be prevented somehow internally in esp-idf. If I find any more data, i'll update.

liuzfesp, 2019-06-06:
Hi @piltom, it's our bug, we will fix it recently.

liuzfesp, 2019-06-06:
HI @piltom I think it should be a bug, we will fix it and update you.

piltom, 2019-07-05:
> 
> 
> HI @piltom I think it should be a bug, we will fix it and update you.

Is there any news?

liuzfesp, 2019-07-14:
For the error log:

> tcpip_adapter: handle_ap_stop 134 esp_wifi_internal_reg_rxcb ret=0x3014

Hopefully the fix for this error message will be merged into master. However, the error log is NOT related to the task watchdog, I guess the task watchdog is caused by other reason. Currently I just can't replicate the task watchdog issue described above. Is the task watchdog easy  reproduced on your side, or it's very difficult to be reproduced?

liuzfesp, 2019-07-14:
@piltom, could you help to do following test:

- Enable task watchdog panic via menuconfig: make menuconfig => Component Config => Common ESP-related => Enable "Invoke panic handler on Task Watchdog timeout" 
- Build the image and reproduce the task watchdog issue
- Get backtrace via "xtensa-esp32-elf-addr2line -piaf -e build/*.elf backtrace_log" to get the backtrace that causes taskwatchdog.

piltom, 2019-07-18:
> For the error log:
>
>
> Hopefully the fix for this error message will be merged into master. However, the error log is NOT related to the task watchdog, I guess the task watchdog is caused by other [...]

It gets triggered all the time. tiT got stuck in an infinite loop, probably due to the wifi error. Unfortunately it's been a month since i posted the issue and i don't have the code anymore.

torresnt, 2019-09-04:
Hello,
I am having the same issue:
********************************************************************************
I (96357) wifi: station: b8:ee:65:bc:99:b7 leave, AID = 1, bss_flags is 134243
I (96357) wifi: new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
main: (152012) event got: 0x00000000
sys: (152012) Processing event, type: 0, id: 17
sys: (152012) SYSTEM_EVENT
sys: (152012) AP_STADISCONNECTED
server: (151956) Stopping Server
I (96497) Device HTTP: Server Stop
wifi: (161768) Disconnecting AP 
server: AP disconnected
sys: (161876) Connecting to: xxxx with password: yyyy
wifi: (161876) Setting WiFi configuration xxxx
wifi: (161876) wifi sta initialized
E (96527) tcpip_adapter: handle_ap_stop 134 esp_wifi_internal_reg_rxcb ret=0x3014
I (96537) wifi: flush txq
I (96537) wifi: stop sw txq
I (96537) wifi: lmac stop hw txq
I (96617) wifi: mode : sta (30:ae:a4:8f:cc:b4)
*****************************************************************************************
Any updates on it? It happens when my device, acting as a server (AP), is connected to a station (Web browser) in APSTA mode and I issue an esp_wifi_stop to restore the connection and connect my device in STA mode to an existing AP or router.

liuzfesp, 2019-11-05:
Hi @piltom @torresnt this issue has been fixed:

- v3.3 - 9a1b11c8
- master - b7d6aa74
- v3.2 - c119a1a0
- v3.1 - 896d597a

Strongly recommend you to upgrade your branch to latest stable IDF releases.

Alvin1Zhang, 2019-11-15:
@piltom Thanks for reporting, feel free to reopen if the issue still happens. Thanks.

closed 2019-11-15 by Alvin1Zhang

Amey112, 2021-09-28:
Hey, I am a beginner to ESP32 and I am using PlatformIO for development, I am getting the exact same problem using:
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Do you know of any workarounds?