espressif/esp-idf #3588: Unknown CMake command "idf_link_components".

tntclaus, 2019-06-05:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      `ESP32-Wrover-Kit`
- Module or chip used:  `ESP32-WROOM-32D`
- IDF version: `v4.0-dev-722-g045aaf6fb`
- Build System:        `CMake`
- Compiler version: `1.22.0-80-g6c4433a`
- Operating System:     macOS
- Power Supply:         `USB`

## Problem Description

I used to compile another version of my firmware with following CMakeLists.txt code:
```
    add_library(renamed_main.elf renamed_main/main.cpp)
    idf_link_components(renamed_main.elf "${components}")
```

But after todays git pull it looks like `idf_link_components` function no longer exists in the toolchain. I checked docs briefly but didn't find anything on this. 

What should I use as a replacement?

renzbagaporo, 2019-06-06:
Hi @tntclaus. We've revamped using ESP-IDF as a library for 4.0 to allow for more flexibility with linking component libraries with external executables/libraries. The API is documented at https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/build-system-cmake.rst#esp-idf-cmake-build-system-api.

The example `build_system/cmake/idf_as_lib` has also been updated to reflect this change.

tntclaus, 2019-06-06:
@renzbagaporo thank you. I figured out what to do and it really seems to be more flexible now. 

For those who want to build multiple versions of the same project here is what you can add to CMakeLists.txt:
```
add_executable(renamed_main.elf renamed_main/main.cpp)
target_link_libraries(renamed_main.elf "${components}")
```

closed 2019-06-06 by tntclaus