espressif/esp-idf #3590: Where can I find GPIO Matrix and how do I modify it to remap pins ?

GeorgeFlorian, 2019-06-05:
OS: Linux Mint 19.1
Board: ESP32 Wrover-B DevKitV4

Hello !

I've just learned that I can remap the GPIOs. But I have no idea how.
I have a ESP32 Wrover-B DevKitV4 that has GPIO 16 and 17, meaning RX2 and TX2, NOT CONNECTED. But I need to use them to connect a GPS Module.

I couldn't find any documentation on where is the GPIO Matrix or how to use it and that's why I've decided to post an issue here.

I feel like I should mention that I've never used ESP-IDF. I used Arduino IDE, but I've since moved to PlatformIO IDE.

negativekelvin, 2019-06-05:
Please don't ignore guidelines, GitHub issues is for software bugs:

> If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

GeorgeFlorian, 2019-06-05:
I've already posted there but I got no answer. If there's no documentation,
what am I supposed to do?

negativekelvin, 2019-06-05:
Be patient for an answer, Google it, read kolban's book.  Read all the pages and pdfs at https://docs.espressif.com/projects/esp-idf/en/latest/

GeorgeFlorian, 2019-06-05:
> Be patient for an answer, Google it, read kolban's book. Read all the pages and pdfs at https://docs.espressif.com/projects/esp-idf/en/latest/

I don't really have the time. It's faster to buy an Wroom module that has RX2 and TX2 connected.

arsab, 2019-06-05:
Use HWSerial where you define your TX, RX:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp

GeorgeFlorian, 2019-06-06:
> Use HWSerial where you define your TX, RX:
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp
> 
> And you should have your details at:
> https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/modules-and-boards.html#esp-modules-and-boards-esp32-wrover

Great reads.

So, by my understanding, I only need to do the following to be able to use RX2 and TX2 on different pins ?
```
#define RXD2 26
#define TXD2 27

void setup() {
	Serial.begin(115200);
	Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
}
```

No need to change something in any of the header files ?

GeorgeFlorian, 2019-06-19:
That's how you actually do it !
```
#define RXD2 26
#define TXD2 27

void setup() {
	Serial.begin(115200);
	Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
}
```

closed 2019-06-19 by GeorgeFlorian