espressif/esp-idf #3591: spiram_psram compilation error

jeromeDms, 2019-06-05:
Dear all
I just updated esp-idf to the latest version and my project no longer compiles.
esp-idf version is v4.0-dev-728

I'm facing the following errors, and cannot find anything useful after searching.

CC build/esp32/spiram_psram.o
/esp32Dev/esp-idf/components/esp32/spiram_psram.c: In function 'psram_enable':
/esp32Dev/esp-idf/components/esp32/spiram_psram.c:577:37: error: 'psram_io.psram_cs_io' may be used uninitialized in this function [-Werror=maybe-uninitialized]
PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[psram_io->psram_cs_io], PIN_FUNC_GPIO);
^
/esp32Dev/esp-idf/components/esp32/spiram_psram.c:617:16: note: 'psram_io.psram_cs_io' was declared here
psram_io_t psram_io;
^
/esp32Dev/esp-idf/components/esp32/spiram_psram.c:750:9: error: 'psram_io.psram_clk_io' may be used uninitialized in this function [-Werror=maybe-uninitialized]
gpio_matrix_out(psram_io.psram_clk_io, SPICLK_OUT_IDX, 0, 0);
^
/esp32Dev/esp-idf/components/esp32/spiram_psram.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
cc1: some warnings being treated as errors
make[1]: *** [spiram_psram.o] Error 1
make: *** [component-esp32-build] Error 2

Anybody here having the same issue ?
Thanks to all
Jerome

Spritetm, 2019-06-06:
I saw that issue as well, I'll put a MR in with the fix. If you need a fix right now, in components/esp32/spiram_psram.c , in the function psram_enable, line 619, change

    psram_io_t psram_io;

into

    psram_io_t psram_io={0};

and the issue should be gone

jeromeDms, 2019-06-06:
Thanks, that did solve :-)

closed 2019-06-10 by igrr