espressif/esp-idf #3592: Wifi - meaning of log message

PerMalmberg, 2019-06-06:
Sorry for the cross-post - I just remembered that I was told to post issues here instead on the sub projects as they are less monitored. I'll close the [other issue](https://github.com/espressif/esp32-wifi-lib/issues/19).

I'm trying to figure out why the wifi driver is running out of memory, when there is 4MB free:

W (115021) wifi: alloc eb len=24 type=3 fail, heap:4098144

I can't find where this is logged so I assume that it is part of the still closed-source libraries(?)

What does type=3 mean in this case and what other possible values are there? Desired type DRAM/IRAM or memory capability?

PerMalmberg, 2019-06-06:
Since updating to the latest master as of today, I'm no longer getting this issue. I can't find anything in the recent commit history that relates to this - thoughts?

liuzfesp, 2019-06-30:
@PerMalmberg type 3 means the WiFi management buffer. The debug log indicates that the WiFi management buffer allocation fails. 

We didn't have any fixes related to WiFi management buffer.

PerMalmberg, 2019-06-30:
@liuzfesp Thanks. What about the two remaining questions?

- Why would this fail when there are 4Mb available?
- What other type values can be logged and what do they mean?

liuzfesp, 2019-07-25:
Hi @PerMalmberg, 
Why would this fail when there are 4Mb available?
WiFi management buffer is allocated from internal memory, although the total memory is 4Mb (internal/external memory), the internal memory may not be enough.

What other type values can be logged and what do they mean?
1 - TX buffer
2 - Long management buffer
3 - Short management buffer
4 - reserved
5 - reserved
6 - RX buffer

PerMalmberg, 2019-07-25:
Is there a way to make these memory buffers use PSRAM?

liuzfesp, 2019-07-26:
HI @PerMalmberg, for memory type 1/2/3, they are used by hardware for TX,  only the internal memory can be used for WiFi TX by hardware.

PerMalmberg, 2019-07-26:
Ok, but why is the log message including the entire heap size and not just memory with the `MALLOC_CAP_INTERNAL` capability? It is confusing as-is.

Also, just how much memory is needed for the management buffers? Anyway to calculate and reserve that?

PerMalmberg, 2019-07-26:
Also, please help me explain the following - there seems to be plenty of both 8-bit and 32-bit alignable memory available compared to the few bytes it tries to allocate so why does the allocation fail?. Legend: F:<free> LB:<largest block> M:<minimum free>
```

I (437871) MemStat: [INTERNAL]
I (437879) MemStat: 8-bit F:33892 LB:32664 M:33236 | 32-bit: F:47728 LB:32664 M:47052
I (437880) MemStat: [SPIRAM]
I (437903) MemStat: 8-bit F:3872088 LB:3867876 M:3864996 | 32-bit: F:3872088 LB:3867876 M:3864996
I (437904) MemStat: Name	Stack	Min free stack
I (437909) MemStat: MainTask	25600	16300
I (437914) MemStat: SocketDispatcher	20480	11984
I (438166) SocketDispatcher: Active sockets: 3
I (442102) wifi: bcn_timout,ap_probe_send_start
W (442103) wifi: alloc eb len=76 type=2 fail, heap:3873224

W (442104) wifi: m f probe req l=0

W (442605) wifi: alloc eb len=76 type=2 fail, heap:3873224

W (442605) wifi: m f probe req l=0

I (442918) MemStat: [INTERNAL]
I (442927) MemStat: 8-bit F:33892 LB:32664 M:33236 | 32-bit: F:47728 LB:32664 M:47052
I (442928) MemStat: [SPIRAM]
I (442951) MemStat: 8-bit F:3872088 LB:3867876 M:3864996 | 32-bit: F:3872116 LB:3867876 M:3864996
I (442953) MemStat: Name	Stack	Min free stack
I (442958) MemStat: MainTask	25600	16300
I (442963) MemStat: SocketDispatcher	20480	11984
W (443106) wifi: alloc eb len=76 type=2 fail, heap:3873320

W (443107) wifi: m f probe req l=0

W (443608) wifi: alloc eb len=76 type=2 fail, heap:3873220

W (443608) wifi: m f probe req l=0

W (444109) wifi: alloc eb len=76 type=2 fail, heap:3873220

W (444109) wifi: m f probe req l=0

```

negativekelvin, 2019-07-26:
It is MALLOC_CAP_8BIT but is it MALLOC_CAP_DMA?

PerMalmberg, 2019-07-27:
Well, I didn't have that MALLOC_CAP_DMA in the bit mask, but even with it set, there seems to be enough memory when allocation fails:

```
I (231637) MemStat: [INTERNAL]
I (231643) MemStat: 8-bit F:35888 LB:32664 M:34760 | 32-bit: F:49724 LB:32664 M:48576
I (231644) MemStat: [INTERNAL & DMA]
I (231649) MemStat: 8-bit F:35888 LB:32664 M:34760 | 32-bit: F:35888 LB:32664 M:34760
I (231654) MemStat: [SPIRAM]
I (231662) MemStat: 8-bit F:3761128 LB:3736232 M:3753204 | 32-bit: F:3761128 LB:3736232 M:3753204
I (231667) MemStat: Name	Stack	Min free stack
I (231674) MemStat: MainTask	25600	16364
I (231678) MemStat: SocketDispatcher	20480	3392
I (232447) ServerSocket: Connection accepted
W (235184) wifi: alloc eb len=24 type=3 fail, heap:3726272

W (235184) wifi: m f null

I (236683) MemStat: [INTERNAL]
I (236693) MemStat: 8-bit F:35304 LB:32664 M:34548 | 32-bit: F:49140 LB:32664 M:48364
I (236694) MemStat: [INTERNAL & DMA]
I (236701) MemStat: 8-bit F:35304 LB:32664 M:34548 | 32-bit: F:35304 LB:32664 M:34548
I (236704) MemStat: [SPIRAM]
I (236713) MemStat: 8-bit F:3711340 LB:3702760 M:3709632 | 32-bit: F:3711340 LB:3702760 M:3709632
I (236717) MemStat: Name	Stack	Min free stack
I (236724) MemStat: MainTask	25600	16364
I (236728) MemStat: SocketDispatcher	20480	3392
W (237393) wifi: alloc eb len=24 type=3 fail, heap:3719836

W (237394) wifi: m f null

I (241733) MemStat: [INTERNAL]
I (241740) MemStat: 8-bit F:34828 LB:32664 M:34284 | 32-bit: F:48664 LB:32664 M:48100
I (241741) MemStat: [INTERNAL & DMA]
I (241747) MemStat: 8-bit F:34832 LB:32664 M:34284 | 32-bit: F:34832 LB:32664 M:34284
I (241751) MemStat: [SPIRAM]
I (241761) MemStat: 8-bit F:3695716 LB:3691092 M:3694128 | 32-bit: F:3696108 LB:3691092 M:3694128
```

negativekelvin, 2019-07-27:
There is a block 32664 which could be SPIRAM_MALLOC_RESERVE_INTERNAL and isn't being used properly?

PerMalmberg, 2019-07-27:
Or, it is all used up? I have no idea how to tell if that is the case, and since this is all in the binary blobs, there's no source to look at. 

My SPIRAM_MALLOC_RESERVE_INTERNAL is set to 32768, the number 32664 in the log is the largest available free block if that is what you're reffering to, @negativekelvin. Or do you mean that these allocations are limited to the memory defined by SPIRAM_MALLOC_RESERVE_INTERNAL and allocated by `esp_spiram_reserve_dma_pool`? I suppose that is a possibility.

@liuzfesp, Can you please clarify from where the WiFi management buffers are allocated and if they in fact are related to SPIRAM_MALLOC_RESERVE_INTERNAL? If they are, how do we calculate the correct amount for SPIRAM_MALLOC_RESERVE_INTERNAL?

negativekelvin, 2019-07-27:
Yes I mean it is suspicious that 32768 and 32664 are so close such that if for some reason that reserved block is not getting used then there may not be another block large enough in the remaining fragmented 2k.  Although 24 bytes is really small it is hard to believe.

PerMalmberg, 2019-08-03:
@liuzfesp any thought on this matter?

liuzfesp, 2019-08-03:
@PerMalmberg, we just found a bug about WiFi/LWIP using PSRAM, it may be related to this issue, will let you know once the fix is merged.

PerMalmberg, 2019-08-08:
@liuzfesp Any progress on fixing/merging the issue? This is preventing me from doing pretty much anything with my device.

liuzfesp, 2019-08-09:
Hi @PerMalmberg fix on master branch: 40e808c6, but it may need few days to automatically sync to github. For other IDF releases, the code has been merged yet.

PerMalmberg, 2019-08-09:
Great news, thank you.

PerMalmberg, 2019-08-17:
@liuzfesp I can see the change via the link, but it doesn't seem to be merged to master yet? (I'm still experiencing the problem on v4.0-dev-1443-g39f090a4f)

PerMalmberg, 2019-08-21:
@liuzfesp I'm sorry to say that even with the update lwip the error persists:

```
W (78383) wifi: alloc eb len=24 type=3 fail, heap:4090072

W (78384) wifi: m f null
```

After this the no sockets are functional.

liuzfesp, 2019-08-22:
Hi @PerMalmberg could you provide your sdkconfig please?

PerMalmberg, 2019-08-22:
@liuzfesp, this is the one:
https://github.com/PerMalmberg/IO-Card-G3/blob/master/software/sdkconfig

liuzfesp, 2019-08-22:
@PerMalmberg your sdkconfig looks ok, I suspect that it's caused by your application memory usage. How much internal memory does the application use? How many tasks the the application create? What's the task stack size of each task?

negativekelvin, 2019-08-22:
@liuzfesp can you explain how allocation of a 24 byte buffer fails when there is 35k free dma heap?  What are the other reasons for failure besides nomem?

```
W (235184) wifi: alloc eb len=24 type=3 fail, heap:3726272
I (236694) MemStat: [INTERNAL & DMA]
I (236701) MemStat: 8-bit F:35304 LB:32664 M:34548 | 32-bit: F:35304 LB:32664 M:34548
```

PerMalmberg, 2019-08-22:
@liuzfesp The memory numbers that @negativekelvin gave you in the post above represents a typical run. I don't see how it matters, but there are four application tasks. I don't remember their stack sizes, but there's 35k DMA-capable available so allocating 24 bytes can't be the actual issue.

Also, the wifi-log shows the number of available PSRAM - and there's loads of memory available there. Is it trying to allocate memory in DMA-capable or och PSRAM?

As much as I would like to dig deeper, it's not possible to since this is one of the binary blobs we don't have the source for. So we're entirely depending on Espressif to help out. Needless to say, a non-functioning Wifi on an ESP32 is a big deal as it's such a vital port of the platform. 

As such, I'm asking in the most possibly kind way - can we please have some focused attention to this matter?

liuzfesp, 2019-08-22:
@negativekelvin @PerMalmberg I just can't reproduce this issue in our auto-test and release test, including throughput and traffic stress test. I wonder is it possible to give me your demo code that can reproduce this issue, then I can debug it directly.

PerMalmberg, 2019-08-22:
@liuzfesp 

Devkit v3.x (or any compatible device), with latest IDF and gcc8.

- Clone my library: https://github.com/PerMalmberg/Smooth
- Change  [this line](https://github.com/PerMalmberg/Smooth/blob/master/CMakeLists.txt#L23) to read `set(selected_test_project http_server_test)`.
- Copy the entire `test/http_server_test/web_root` folder (including the folder itself) to the root of an SDCard and put that into the device.
- Compile and flash the device. 
- Direct your browser to http://ip-of-device:8080

Sometimes it happens on the page-load try, sometimes it takes a few reloads but it happens very consistently.

PerMalmberg, 2019-08-22:
@liuzfesp Sorry, forgot to mention that you'll have to set up the Wifi credentials in `wifi_creds.h` too.

PerMalmberg, 2019-08-22:
Here's an output from a run I just did. As you can see, there has never been less than 24908 bytes DMA-capable memory available. It's not a huge amount, but it is certainly more than the 24 bytes it tries to allocate.

And, talking about memory types - we've never gotten an answer on the question from where the memory is requested. The log displays the available heap - is that from where it is trying to allocate?

Let me know if there's any more information I can provide.

```
I (43020) HTTPServer: Reply: OK
I (43041) HTTPServer: Request: GET: '/img/application_view_icons.png'
W (43048) wifi: alloc eb len=24 type=3 fail, heap:4066848

W (43049) wifi: m f null

I (43076) HTTPServer: Reply: OK
I (43099) HTTPServer: Request: GET: '/img/application_view_list.png'
I (43133) HTTPServer: Reply: OK
I (43151) HTTPServer: Request: GET: '/img/application_view_tile.png'
W (43154) wifi: alloc eb len=24 type=3 fail, heap:4066964

W (43155) wifi: m f null

W (43156) wifi: alloc eb len=24 type=3 fail, heap:4065408

W (43160) wifi: m f null

I (43182) HTTPServer: Reply: OK
I (43198) HTTPServer: Request: GET: '/img/application_xp.png'
I (43231) HTTPServer: Reply: OK
I (43248) HTTPServer: Request: GET: '/img/application_xp_terminal.png'
I (43272) HTTPServer: Reply: OK
I (43289) HTTPServer: Request: GET: '/img/arrow_branch.png'
I (43313) HTTPServer: Reply: OK
I (43334) HTTPServer: Request: GET: '/img/arrow_divide.png'
I (43358) HTTPServer: Reply: OK
I (46890) SocketDispatcher: Active sockets: 8
W (48160) SocketDispatcher: Receive timeout on socket 0x3fffe71c (5000 ms)
I (48162) Socket: [, 0, 56, 0x3fffe71c]: Receive timeout
I (48164) Socket: [, 0, 56, 0x3fffe71c]: Socket stopping
I (48185) Socket: [, 0, -1, 0x3fffe71c]: Disconnected
W (48202) SocketDispatcher: Receive timeout on socket 0x3ffe3dfc (5000 ms)
I (48204) Socket: [, 0, 58, 0x3ffe3dfc]: Receive timeout
I (48206) Socket: [, 0, 58, 0x3ffe3dfc]: Socket stopping
I (48226) MemStat: [INTERNAL]
I (48232) Socket: [, 0, -1, 0x3ffe3dfc]: Disconnected
I (48232) MemStat: 8-bit F:31736 LB:24952 M:24908 | 32-bit: F:61944 LB:30208 M:55108
I (48236) MemStat: [INTERNAL | DMA]
I (48242) MemStat: 8-bit F:31740 LB:24952 M:24908 | 32-bit: F:31740 LB:24952 M:24908
I (48249) MemStat: [SPIRAM]
I (48256) MemStat: 8-bit F:4057176 LB:4050444 M:3957572 | 32-bit: F:4057176 LB:4050444 M:3957572

```

liuzfesp, 2019-08-23:
@PerMalmberg we are now debugging this issue, will update your once we have some progress

Rhulin, 2019-08-26:
@PerMalmberg Is there a way to reproduce the problem without using SD card?

PerMalmberg, 2019-08-26:
@Rhulin I don't know, I haven't tried. Why do you ask?

PerMalmberg, 2019-08-28:
@Rhulin run "flash" instead of "app-flash" to write a new partition table (assuming you have it configured). Are you debugging this issue or just trying out my library? If the later, then this isn't the place to ask for help for that.

@liuzfesp Any progress? A "yes" or "no" will do fine for now, thank you :)

PerMalmberg, 2019-08-28:
@Rhulin Ok. Unless you're working for Espressif, there's the small issue that we don't have access to the wifi source code to debug this - at this time it appears the problem is likely located in the closed-source part of IDF.  You're of course more than welcome to see if you can find anything, but you should at least know this.

Rhulin, 2019-08-28:
Hi @PerMalmberg How can I solve this issue                   

E (761) esp_image: Image length 1672160 doesn't fit in partition length 1048576
E (761) boot: Factory app partition is not bootable
E (763) boot: No bootable app partitions in the partition table

Rhulin, 2019-08-28:
Hi @PerMalmberg I am trying to reproduce the problem and debug

negativekelvin, 2019-08-29:
https://github.com/espressif/esp-idf/issues/2892#issuecomment-526268423

PerMalmberg, 2019-09-06:
@liuzfesp Is it too early to ask for a progress update?

PerMalmberg, 2019-09-10:
@liuzfesp Could we please have an update?

I'm currently developing the web UI for my product but trying to test-run it against the ESP32 is near to impossible since the Wifi constantly breaks and requires a restart of the device.

PerMalmberg, 2019-09-10:
Not sure if it helps, but I just got this in the log; I can't remembering ever seeing type 2 failing before:

```
W (5217413) wifi: alloc eb len=76 type=2 fail, heap:3032912
```

Memory stats when it happened; F:(free) LB:(largest block): M:(minimum free)
```
I (5220816) MemStat: [INTERNAL]
I (5220894) MemStat: 8-bit F:20400 LB:17596 M:17904 | 32-bit: F:42436 LB:22036 M:39920
I (5220908) MemStat: [INTERNAL | DMA]
I (5220958) MemStat: 8-bit F:20424 LB:17596 M:17904 | 32-bit: F:20424 LB:17596 M:17904
I (5220974) MemStat: [SPIRAM]
I (5221084) MemStat: 8-bit F:3028588 LB:2941692 M:3026568 | 32-bit: F:3028684 LB:2941692 M:3026568
```

PerMalmberg, 2019-09-11:
@projectgus Sorry to ping you, but @liuzfesp seems to have gone radio silent. Has any progress been made since the last statement from @liuzfesp 20 days ago?

Also, shouldn't this issue be on the list of known issues in IDF v.4.0-beta 1 which you announced earlier today? (and the PSRAM cache issue too, since v4.0 uses gcc 8.2?)

projectgus, 2019-09-12:
HI @PerMalmberg,

@Rhulin has been debugging this and will update with any progress.

Some other ideas from looking at the case:

I notice that some time passes between the "MemStat" dumps and the errors. Are you able to turn on [memory tracing (of all allocations)](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html#heap-tracing) in between MemStat calls? You should be able to store the trace buffer in PSRAM provided nothing is trying to allocate while writing/erasing flash.

Also, could you please try enabling "Comprehensive" heap poisoning and running a `heap_caps_check_integrity(MALLOC_CAP_DMA)` at some point after the error, to make sure the heap metadata is correct?

> Also, shouldn't this issue be on the list of known issues in IDF v.4.0-beta 1 which you announced earlier today? (and the PSRAM cache issue too, since v4.0 uses gcc 8.2?)

We've assumed that issues open on GitHub don't need to be mentioned here, especially if the root cause isn't confirmed (until we can reproduce without your code involved, it's not certain that this is an issue in IDF). But you're right we should at least mention this in the known issues, and link to major issues like the PSRAM cache one. Will fix for the next pre-release.

PerMalmberg, 2019-09-12:
Ah, so @Rhulin is working for Espressif, it wasn't obvious from the questions s/he asked, my apologies for that.

Yes, I can and will do what you ask, @projectgus but isn't it already clear that this happens even without *my* code; there are already three other issues linked by @negativekelvin :  #3976,  #2184  and #3840, all of them showing the same symptoms.

projectgus, 2019-09-12:
> Yes, I can and will do what you ask, @projectgus but isn't it already clear that this happens even without my code; there are already three other issues linked by @negativekelvin : #3976, #2184 and #3840, all of them showing the same symptoms.

Apologies, you're right and I should have looked through the full links.

Am I right in summarising that the root causes mentioned in the other issues are either the PSRAM bug with confirmed toolchain fix, or that running out of internal DMA memory causes things not to recover when memory frees up? Do you think one of these might be the root cause here, as well?

PerMalmberg, 2019-09-12:
It is of course possible that the PSRAM cache issue is the underlying cause for this issue, at least if memory in fact is being allocated from PSRAM. Two of the linked issues uses gcc 8 and idf-master, i.e. v.4x. The third one uses commit 52f9a5c from April 24th 2018 so that's likely gcc 5. Since fixes for gcc 8 are not available, only Espressif can say anything about whether or not the PSRAM cache is the cause.

Regarding running out of DMA memory - I'd say that is not the case. Based on the memory statistics there have always been ~20ish kB available (all-time minimum free), and the logs complain about failing to allocate <30b so there obviously were enough memory to allocate at the time.

Btw, I'm a bit confused re. the relation between DMA memory/PSRAM and Wifi. On failure the Wifi log displays the amount of PSRAM available, implying that it tries to allocate from PSRAM. Is it really, or is it misleading information?

PerMalmberg, 2019-10-19:
I was never able to make heads from tails of the trace and since I've not received a reply in more than a month, I did another small test - I maxed out  the amount of reserved DMA memory (262144), reduced CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL to 64 and configured the system to run in unicore mode to avoid any issues with #2892. Also, SPIRAM_TRY_ALLOCATE_WIFI_LWIP is enabled.

What now happens is that wifi-allocations fails immediately at start up, which is unexpected since we now have 
- Maximum amount of reserved DMA memory
- 4MB or (presumably) usable SPIRAM

```
I (5911) wifi: state: auth -> assoc (0)
I (5918) wifi: state: assoc -> run (10)
I (5935) wifi: connected with Our place, channel 1, 40U, bssid = f0:9f:c2:aa:b0:3f
I (5936) wifi: pm start, type: 1

W (5987) wifi: alloc eb len=24 type=3 fail, heap:4093504

W (5987) wifi: m f null

I (6467) SocketDispatcher: Active sockets: 0
W (6478) wifi: alloc eb len=24 type=3 fail, heap:4055916

W (6478) wifi: m f null

I (6505) Stora hallen: Triggered.
I (12627) MemStat: Mem type |  8-bit free | Smallest block | Minimum free | 32-bit free | Smallest block | Minimum free
I (12630) MemStat: INTERNAL |      129240 |         127492 |       128704 |      191320 |         127492 |       190764
I (12641) MemStat:      DMA |      129240 |         127492 |       128704 |      129240 |         127492 |       128704
I (12654) MemStat:   SPIRAM |     4056312 |        4055140 |      4052560 |     4056312 |        4055140 |      4052560

```
As you can see, there has never been less than 128704 of DMA available - I'd say this proves that there are no huge buffers consuming the DMA and preventing wifi-allocations from succeeding. 

This question is now why does the wifi-allocation fail even when there is such a large amount of DMA available?

@projectgus @Rhulin Have you done any new findings?

negativekelvin, 2019-10-20:
When the memory gets reserved, it loses the MALLOC_CAP_DEFAULT attribute flag.  If the wifi driver is using this flag then it won't match the reserved pool?

negativekelvin, 2019-10-20:
https://github.com/espressif/esp-idf/blob/51b4e97e4255da824ccb45a65b25d97b689ce0cb/components/esp32/esp_adapter.c#L410-L432

negativekelvin, 2019-10-20:
Also affecting Bluetooth

https://github.com/espressif/esp-idf/blob/51b4e97e4255da824ccb45a65b25d97b689ce0cb/components/bt/controller/bt.c#L772-L775

negativekelvin, 2019-10-21:
So does changing those functions to use `MALLOC_CAP_8BIT|MALLOC_CAP_DMA|MALLOC_CAP_INTERNAL` solve it for you, @PerMalmberg ?

PerMalmberg, 2019-10-21:
Theoretically it probably will, though I haven't tried. I'd like to have some feedback on what @Rhulin and @projectgus have found before I start messing around with the inner workings of IDF.

While you have a good point, @negativekelvin , is it really the right solution? I don't know, I don't know the reasoning behind the current design so I can't say if it is the right way to go or not.

negativekelvin, 2019-10-21:
Seems obvious, it is exactly what I suggested at the beginning. Do we want wifi to have access to the reserved pool? Yes, of course. Then we have to change the flags. It will still try to use the other unreserved heaps first because they are matching MALLOC_CAP_8BIT so it should be exactly what we want.

PerMalmberg, 2019-10-22:
Updating to use `MALLOC_CAP_8BIT|MALLOC_CAP_DMA|MALLOC_CAP_INTERNAL`  does seem to solve the problem.

falkoschindler, 2019-10-23:
> Updating to use `MALLOC_CAP_8BIT|MALLOC_CAP_DMA|MALLOC_CAP_INTERNAL` does seem to solve the problem.

I think I'm having the very same problem, but this solution did not help. It just seems to postpone the error. Before my application failed after 10 minutes, now it was alive for 19 minutes.

negativekelvin, 2019-10-23:
@falkoschindler well you're still going to have to tune SPIRAM_MALLOC_RESERVE_INTERNAL and other parameters to fit your application and make sure there are no memory leaks.  There is a physical limit on ram so it will always be possible to run out.

PerMalmberg, 2019-10-23:
Did some further testing today and I can no longer cause this issue to occur.

So how quickly can we get the proposed changes into master, @projectgus and @Rhulin  ?

projectgus, 2019-10-25:
Hi @PerMalmberg ,

The fix suggested by @negativekelvin looks correct to me, thanks for persisting with this. Feel free to send a PR, although we can also make the change internally.

Due to some other major changes coming down the pipeline, we may not be able to merge this for a short time.

Angus

PerMalmberg, 2019-10-25:
thanks for the update, @projectgus. Can you share some information on what those major changes are? Is it something we can/need to prepare for?

projectgus, 2019-10-28:
@PerMalmberg We're merging ESP32-S2 Beta support onto master. The intention is for no impact on ESP32 users, but there may be some unexpected regressions. Suggest moving to release/v4.0 branch if you need more stability than master.

filzek, 2019-10-29:
@negativekelvin. Today we updated the SDK with git pull --rec and it has fixed some errors on iram0, but the size needed in flash has increased more than 200k so something still not right. But iram0 fits now ok and a lot of components stop using a lot of iram

[25 more replies]

closed 2022-05-17 by jack0c