espressif/esp-idf #3594: "spi_device_transmit" costs much more time

BigSmileFace, 2019-06-06:
Hi
in spi master mode,  we use "spi_device_transmit" to start a spi transmissionfrom .
and we found it spends about 46us to send 1 byte.
from the datasheet [https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html](url)，we know normally it is 26us (with dma and interrupt)
we calculate the time with function"esp_timer_get_time()".

version is release 3.3
`

Frequency (MHz) | Transaction Interval (us) | Transaction Length (bytes) | Total Time (us) | Total Speed (kBps)
-- | -- | -- | -- | --
8 | 25 | 1 | 26 | 38.5
8 | 25 | 8 | 33 | 242.4
8 | 25 | 16 | 41 | 490.2
8 | 25 | 64 | 89 | 719.1
8 | 25 | 128 | 153 | 836.6

`

so could u give us some advice adout how  to improve the speed?Thanks!!

code:
`		spi_bus_config_t buscfg={
			.miso_io_num=PIN_NUM_MISO,
			.mosi_io_num=PIN_NUM_MOSI,
			.sclk_io_num=PIN_NUM_CLK,
			.quadwp_io_num=-1,
			.quadhd_io_num=-1
		};
		spi_device_interface_config_t devcfg={
			.cs_ena_pretrans = 0,
			.cs_ena_posttrans = 10,
			.clock_speed_hz=SPI_SPEED_HZ,           
			.mode=3,                                
			.spics_io_num=PIN_NUM_CS,              
			.queue_size=1,                          
			.pre_cb=NULL,  									
	};
	
	ret=spi_bus_initialize(HSPI_HOST, &buscfg, 2); //set 0 means no dma
		
	ret=spi_bus_add_device(HSPI_HOST, &devcfg, &spi);

	//malloc in dma memory
	spiTxBuf = (uint8_t *)heap_caps_malloc(SPI_BUF_MAX_LEN_BYTES, MALLOC_CAP_DMA);
	spiRxBuf = (uint8_t *)heap_caps_malloc(SPI_BUF_MAX_LEN_BYTES, MALLOC_CAP_DMA);`

ginkgm, 2022-02-24:
Hi @BigSmileFace ,
I think you have got one possible answer, the buffer. Do you directly use the buffer from its beginning address, or use something like `spiRxBuf+3`? The RX buffer is required to be aligned, otherwise malloc+free will still happen.

If you still have problem, you may come up with the full code (how you make a transaction descriptor). 

Sorry for late reply.

ginkgm, 2022-06-27:
Closed, feel free to reopen if you see same issue

closed 2022-06-27 by ginkgm