espressif/esp-idf #3595: Could there be an official docker image for esp-idf?

cranphin, 2019-06-06:
Hi!

Docker is mighty useful and more and more popular.
I'm using it to have an easy to update/replace esp-idf on Windows which only requires docker installed.

Is there any chance of having an official docker image for this?

Currently I'm running Travis builds using this: https://github.com/cranphin/esp32-idf-docker
This uploads an image on docker hub: https://hub.docker.com/r/cranphin/esp32-idf
Travis isn't needed, Docker hub has build automation built in, but doesn't support daily build (only builds triggered by code changes).
Mayby this could provide a sample of how to do this :)

igrr, 2019-06-06:
Hi @cranphin, yes, providing esp-idf docker images for release versions is something we plan to do. Thank you for the hint about Docker Hub automated builds. They indeed look more straightforward than the previous idea we had (build in our internal CI and push to the Docker Hub).

dobairoland, 2020-04-28:
The docker file is available here: https://github.com/espressif/esp-idf/tree/master/tools/docker

closed 2020-04-28 by dobairoland

igrr, 2020-04-28:
Also see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html for the overview.

gavril0, 2020-06-22:
Hi, 

I would like to setup an full environment in a container to develop for ESP32 on Windows 10 with Visual Studio Code.   In the container, I would expect to install ESP-IDF SDK and all needed  tools  (gcc, ninja, python, ...)  to avoid conflict with other versions of these tools installed for other projects (e.g. RTools) and to manage Windows PATH env variable.  I don't want/need PlatformIO and prefer to use Epresssif tool sdirectly. 

I would also like to build, flash and debug from VS Code.  I have seen that it is in principal possible to access serial port from windows container:

https://docs.microsoft.com/en-us/virtualization/windowscontainers/deploy-containers/hardware-devices-in-containers

but people working with ESP seems to use WSL2 and Linux containers on Windows and, apparently, it is not straightforward to connect to serial port in this case (see @cranphin above). Ideally, I would like to avoid Linux on Windows if possible. 

- Is there some instruction about how to configure Visual Studio Code ? 
- Should I install VS Code docker extension as indicated [here](https://code.visualstudio.com/docs/containers/overview)?
- Should I install [Expressif Official VScode Extension](https://esp32.com/viewtopic.php?t=13669)?
- Should I install [Epressif docker image](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html)? Is it a Windows or WSL container?

Is there some instructions about the steps? Sorry if some questions might appear naive but I never used containers, ESP32,  VSCode. I am an academic who has done mostly C/C++/WinAPI with Visual Studio and has  limited familiarity with git, linux and most web technologies.

Thanks

P.S. Sorry if I should I have open a new issue rather than left a comment.... I am also new at that.

igrr, 2020-06-23:
Hi @gavril0,

> In the container, I would expect to install ESP-IDF SDK and all needed tools (gcc, ninja, python, ...) to avoid conflict with other versions of these tools installed for other projects (e.g. RTools) and to manage Windows PATH env variable.

Starting from IDF v4.0, it is no longer necessary to modify Windows PATH variable to develop with ESP-IDF. This is true both for the command line tools and for E**s**pressif VS Code extension. I recommend giving this option a try, if you prior experience includes native Windows development. You can get the tools installed using ESP-IDF Tools installer: https://docs.espressif.com/projects/esp-idf/en/stable/get-started/windows-setup.html#esp-idf-tools-installer. (If you would like more information on how the tools are installed, and why the system PATH doesn't need to be modified, please see [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#downloadable-tools)).

> https://docs.microsoft.com/en-us/virtualization/windowscontainers/deploy-containers/hardware-devices-in-containers

This page refers to Windows containers running on Windows. Unfortunately serial ports are still not available to Linux containers running on Windows, see https://github.com/docker/for-win/issues/1018#issuecomment-416707198. There are some possible workarounds, including forwarding the serial port into the container over TCP (RFC2217), but nothing we can really recommend as a good "getting started" option yet.

cranphin, 2020-06-30:
Yeah, I was/am hoping wsl2 will support it at some point :)
In the meanwhile a hacked RFC2217 has been working for me (See https://github.com/cranphin/esp32_win_buildscripts)