espressif/esp-idf #3598: CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_* is broken in menuconfig/genconfig

ducalex, 2019-06-06:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it):  v3.3-beta1-728-g8c57aa024
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is no longer available.

In commit 37126d3451eabb44eeeb48b8e2ee554dc233e2a8 the indentation in   components/spi_flash/Kconfig was messed up and it's now impossible to allow the option.

### Expected Behavior

Have a choice of Aborts/Fails/Allowed

### Actual Behavior

No choice

### Steps to repropduce

1. Make menuconfig
2. Go to Components -> SPI Flash
3. See Writing to dangerous flash regions item

### Code to reproduce this issue

None 

## Debug Logs

None

## Other items if possible

None

dobairoland, 2019-06-11:
Thank you @ducalex for reporting this issue. It will be fixed ASAP.

closed 2019-06-13 by igrr

dobairoland, 2019-06-14:
Please note that this has been fixed in master and will be fixed with a little delay in v3.3 as well. Thanks for your patience.