espressif/esp-idf #3599: ESP32 will not start

boxy85, 2019-06-07:
I cannot get the ESP32 board to run using ESP home no matter what config i try. I have removed all information regarding the camera from the Yaml file and it still doesnt load.

I have used 4 different boardtypes in my yaml file. I have tried just a standard ESP intstall with no other components. Any help would be appreciated

If i install the standard esp32 camera webserver using arduino then the camera works perfectly. I have a esp32-s Ai-thinker board with a camera attached. 
I am using NodeMCU to upload the .bin file.

My yaml file looks like:

```
esphome:
  name: cam2
  platform: ESP32
  board: nodemcu-32s
  
wifi:
  ssid: "redacted"
  password: "redacted"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "redacted"

ota:
  password: "redacted"
# Author by airijia.com
# ESP32-CAM
# Lovelace camera card https://github.com/custom-cards/camera-card
# Video tutorial https://youtu.be/-towjof_ixU

#wifi:
#  ssid: $ssid
#  password: $password
#  fast_connect: True
#api:
#  reboot_timeout: 0s
#ota:
#logger:

# ESP32-CAM

# Example configuration entry
esp32_camera:
  external_clock:
    pin: GPIO21
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO4, GPIO5, GPIO18, GPIO19, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  reset_pin: GPIO15
  resolution: 640x480
  jpeg_quality: 10
  # Image settings
  name: My Camera
# ... Frame Settings

# Flashlight
#output:
#  - platform: gpio
#    pin: GPIO4
#    id: gpio_4
#light:
#  - platform: binary
#    output: gpio_4
#    name: "light"
```

negativekelvin, 2019-06-07:
I think this issue belongs here https://github.com/esphome/issues/issues ??

boxy85, 2019-06-07:
moved to correct area

closed 2019-06-07 by boxy85