espressif/esp-idf #3600: Is it possible to map a nvs blob field into the memory?

sysizlayan, 2019-06-07:
Hi,
Currently, we are storing our CA files in NVS as a multipage blob. After the reset of ESP32, the CA file is read from the NVS to a globally reachable byte array. I wonder if we can map the flash memory to be used as if it is in the SRAM, read-only.

Or, is there a better way to store large CA files to be used for MQTT and SSL communications? I just want to give pointer located in the flash, reducing the memory consumption.

Thanks in advance.

negativekelvin, 2019-06-07:
I don't think so because it is not necessarily stored as a single blob and can be chunked and reordered.

igrr, 2019-06-10:
@sysizlayan As @negativekelvin has explained, the blob may not be stored as a continuous mmap-able region in the NVS. Besides, you probably would want to use encrypted NVS for certificates, but the memory mapping hardware can not transparently decrypt the NVS data, as it is encrypted with a different algorithm than the one used for Flash Encryption.

What you can do is, define a partition with type "data" and a custom subtype in the partition table, write the CA bundle into that partition (at production time, over serial, or at the first boot, using esp_partition_write API), and then call esp_partition_mmap on that partition at run time.

Yet another option is to embed the CA bundle into the application binary (as the certificates are embedded in some of the example). This way, to update the bundle you perform an OTA update of the application. This is the easiest method, and it allows the OTA rollback feature to work for the CA bundle as well. In other methods, the CA bundle format would need to stay backwards-compatible, or else the rollback feature should not be used.

igrr, 2019-11-04:
Closing as the question has been answered.

closed 2019-11-04 by igrr