espressif/esp-idf #3603: MPI (bignum) acceleration breaks mbedTLS with DHE-RSA-AES256-GCM-SHA384

PerMalmberg, 2019-06-09:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version v3
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.0-dev-728-g826ff7186
- Build System:         CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

As first posted on the forum [here](https://www.esp32.com/viewtopic.php?f=2&t=10918&p=44691#p44686) I experienced task hangs when connecting using OpenSSL with DHE-RSA-AES256-GCM-SHA384.

### Expected Behavior

No task hangs

### Actual Behavior

Task hangs when calling mbedtls_ssl_handshake_step

### Steps to repropduce

Enable "MPI (bignum) accelecration" in menuconfig for the https_server example.
Compile, flash and run https_server example

run this command: `openssl s_client -cipher "DHE-RSA-AES256-GCM-SHA384" -connect N.N.N.N:443`

At this point the log will starting emitting these log messages at which point no further connections to the device is possible.

> E (113156) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (113156) task_wdt:  - IDLE1 (CPU 1)
E (113156) task_wdt: Tasks currently running:
E (113156) task_wdt: CPU 0: IDLE0
E (113156) task_wdt: CPU 1: httpd

The config used: [sdkconfig.zip](https://github.com/espressif/esp-idf/files/3269497/sdkconfig.zip)

This issue seems similar to https://github.com/espressif/esp-idf/issues/1681 but the fix for that was committed back in 2018 so this is either a regression, or something else.

projectgus, 2019-06-10:
Hi @PerMalmberg ,

Thanks for the bug report. We ddn't get full confirmation of fixing the bug in #1681 so it's possible it's the same one.

Thanks for the easy steps to reproduce. Will investigate further.

PerMalmberg, 2019-06-11:
@projectgus This might be of interest to you: https://forums.mbed.com/t/solved-mbedtls-ssl-handshake-step-hanging/5179/3

KonstantinKondrashov, 2019-06-12:
Hi @PerMalmberg! Thanks for your bug report. I have investigated this bug and figure out that our HW implementation [mbedtls_mpi_exp_mod()](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/esp_bignum.c#L346) distinguish from [software implementation mbedtls](https://github.com/espressif/mbedtls/blob/19eb57f7f7b2f4312d497ddbcb5f104fc6877c70/library/bignum.c#L1630). In particular the differents in returned structure `mbedtls_mpi *Z` where `Z.p` is ok but `Z.s` and `Z.n` fields diffrent. (The `s` field is not set at all, `n` less by 1).

I am not sure how to fix it correctly but you can try this desicion:
[fix_mbedtls_mpi_exp_mod.zip](https://github.com/espressif/esp-idf/files/3280897/fix_mbedtls_mpi_exp_mod.zip)
`MBEDTLS_MPI_CHK( mbedtls_mpi_grow(Z, m_words) );` is replace by `MBEDTLS_MPI_CHK( mbedtls_mpi_copy(Z, Rinv ) );` [here](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/esp_bignum.c#L395).

Update `mbedtls` to v2.16.1. does not allow ilimunate this issue.

We have opened the MR to update `mbedtls` to v2.16.1.

PerMalmberg, 2019-06-12:
> Update mbedtls to v2.16.1. does not allow ilimunate this issue.
Sorry, what does that mean?

KonstantinKondrashov, 2019-06-12:
I tried to use mbedtls v2.16.1 to check it fixes this problem or not. This issue still exists, it does not depends on the version.
In our implementation of  mbedtls_mpi_exp_mod() need correctly set Z.s and Z.n. I work on it now.

Thanks.

KonstantinKondrashov, 2019-06-14:
Hi @PerMalmberg.
We fixed this bug, it will merge as soon as possible. If this bug blocks you, you can use this patch [5222.zip](https://github.com/espressif/esp-idf/files/3288867/5222.zip)

Thanks.

PerMalmberg, 2019-06-14:
I can wait for it to be merged. Thanks for the quick resolution.

closed 2019-06-26 by igrr