espressif/esp-idf #3606: guru meditation error: core 0 panic'ed (cache disabled but cached memory region accessed) when call outbox_enqueue() in mqtt_outbox.c

zaazbb, 2019-06-10:
guru meditation error: core 0 panic'ed (cache disabled but cached memory region accessed) when call outbox_enqueue() in mqtt_outbox.c.
and when i am saving my own data use  NVS blobs.

[image]

i use esp-idf-v3.2, the mqtt lib was Integrated to esp-idf.

what can i do, to avoid this panic?

david-cermak, 2019-06-10:
Hi @zaazbb 

Can you please include also the missing part of the screen, so that all information from core dump is included?
Also please share the exact version of IDF you use.
Thank you!

Alvin1Zhang, 2019-10-11:
@zaazbb  Thanks for reporting, would you please help share more details of the issue? Thanks.

Alvin1Zhang, 2020-01-16:
@zaazbb Thanks for reporting, would you please help share more details of the issue? Any updates ? Thanks.

Alvin1Zhang, 2020-03-03:
@zaazbb Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details provided. Thanks.

closed 2020-03-03 by Alvin1Zhang