espressif/esp-idf #3607: Use BLE for OTA upgrades

zhanlej, 2019-06-10:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    ``v4.0-dev-621-g81ca1c013``
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I now use the esp32 BLE link in conjunction with our own protocol to transfer OTA files. However, the transmission rate of BLE is very slow, and the bin file size of the device is 700K+. So it takes about 10 minutes to transfer the firmware (chunksize is 512 bytes), I think the time is too long.

So I have a few questions I would like to ask:
1.  Does ESP_IDF have an OTA upgrade API specifically for BLE such as Nordic DFU?
2.  Can the firmware of the device be divided into several parts for OTA upgrade? Just like esp8266.

chegewara, 2019-06-10:
Hi,
from my OTA over BLE tests i did few weeks/months ago i can tell that 10 minutes is way too long. If i remember i could achieve about 500kbits/s with notify and about 110kbits/s with indications, which would give about 1 minute to flash esp32.

I didnt test it, but as long as you can feed `esp_ota_write` with data between calling `esp_ota_begin` and `esp_ota_end` with the same handle, it does not matter how much you divide your bin or even from how many sources you get it.

zhanlej, 2019-06-10:
Hi, @chegewara 
Can you provide your test code?

My scenario is to use the APP as the client and the device as the server, so the device cannot receive the notify. The APP divides the firmware into 512 bytes per packet and sends it to the device using the write command.

I think this way should be faster than using notify/indication. As far as I know, indicicate supports a maximum of 20 bytes per packet. If you want to send more than 20 bytes of data, you need to add the package and unpacking protocol to the data.

chegewara, 2019-06-10:
If you want to use notify/indicate with more than 20 bytes you need to change MTU.

Notify is faster than indication or write request but less reliable, because some packets may be lost.

Yes, if you are using esp32 in server mode then you will have to use write, but still, it does not matter if you are sending data in 512 bytes chunks as long as you dont increase MTU to 517, because your 512 bytes are sent in 22 bytes (if i remember) chunks by API internally. This explain why it takes so long.

zhanlej, 2019-06-11:
Hi,
I learned from this article ([link]) that if you do not use HID, the transfer rate of using iphone6 is only 2,667.66 B/s, how do you achieve it, then 500kbit/s?

[Link]:https://punchthrough.com/pt-blog-post/maximizing-ble-throughput-on-ios-and-android/

chegewara, 2019-06-11:
Ive been doing tests between 2 esp32s, im not using Apple devices.

Weijian-Espressif, 2019-06-17:
@zhanlej Connection interval and MTU will affect throughput. The maximum throughput of two ESP32s can reach 720kbps, you can refer to our ble_throughput example.

closed 2021-03-22 by Alvin1Zhang