espressif/esp-idf #3608: Problems with esp32 compilation

tutuwin, 2019-06-10:
----------------------------- Delete below -----------------------------

/home/linux/esp32/esp-idf/examples/get-started/ESP32_RC522/build/esp32/libesp32.a(cpu_start.o):(.literal.main_task+0x18): undefined reference to `app_main'
/home/linux/esp32/esp-idf/examples/get-started/ESP32_RC522/build/esp32/libesp32.a(cpu_start.o): In function `main_task':
/home/linux/esp32/esp-idf/components/esp32/cpu_start.c:515: undefined reference to `app_main'

chegewara, 2019-06-10:
Do you have `void app_main()` function in any file in your project?
What is yours folders tree looks like?

koobest, 2019-07-02:
If you are using Cmake to build your project, make sure your c source filename is included in `CMakeLists.txt` under the main folder.

```
set(COMPONENT_SRCS "hello_world_main.c"
                                              "file1.c"
                                              "file2..c")
```

f00bard, 2019-07-02:
According to [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#example-project) a CMakeLists.txt file isn't needed under the <project>/main folder.  However I'm running into the same issue (having used GNU Make until now).

renzbagaporo, 2019-07-03:
Hi @f00bard , that part of the docs is outdated, will fix. `main` should be treated like any other component, with its own CMakeLists.txt.

renzbagaporo, 2019-08-12:
@tutuwin Closing as there are no updates to the last question by @koobest.

closed 2019-08-12 by renzbagaporo

Arag0to, 2020-02-06:
Hi @renzbagaporo ,
I am running in the same kind of problem.
[image]

This problem is happening to me because I'm trying to work in c++ with Esp-IDF V4.0 so I renamed the main.c with main.cpp and did the same in the CMakeList.txt.

It work for some time but then the library becomes "unresolved including" and this errors comes with it some time.
I don't totally understand this erros and would like to know of a fix that is good for more then a day.

Thanks

chegewara, 2020-02-06:
Did you add:
`extern "C" void app_main()`

Arag0to, 2020-02-07:
I tried it last week and it did'nt work but this time it did.
But new problem,
### main.cpp
[image]
It's a C main so can't use any C++ stuff.
It's a more basic problem but could someone help me. I found so little information on this and when I did I'm not sure if it's still good or expired since I'm on ESP-IDF V4.0

Also, I don't have the compiler error but since it still compile my code in C, it does'nt recognize the include in the class.
### spi.h
[image]
### spi.cpp
[image]

renzbagaporo, 2020-03-03:
Have you added `spi.cpp` to the list of source files @Arag0to?

su-Koch, 2020-03-30:
> I tried it last week and it did'nt work but this time it did.
> But new problem,
> ### main.cpp
>
> [image]
> It's a C main so can't use any C++ stuff.
> It's a more basic problem but could someone help me. I [...]

I have exactly the same issue. I'trying to add an external class (cpp + h). As I'm a beginner in cmake (coming from Visual Studio), can anyone tell me how to include source and header files correctly to project? Thanks a lot.

A-Babin, 2021-10-12:
Haven't seen this documented anywhere, but I've found that if you're defining your `app_main` in C++, the file containing it must not be named `main.cpp`, even if you do everything else right.

A-Babin, 2021-10-12:
> > Haven't seen this documented anywhere, but I've found that if you're defining your `app_main` in C++, the file containing it must not be named `main.cpp`, even if you do everything else right.
> 
> Almost true. It has to be: `extern "C" app_main()` and can be in `main.cpp`

If that's what's intended, then there's a bug. IDF v4.3.1, Windows 10, just installed yesterday, I have a fresh copy of `hello_world.` Enabled C++ and added `extern "C"` and all that... ninja fails with `undefined reference to `app_main'`. I do nothing except rename `main.cpp` and update the main component CMakeLists, and it builds fine.

chegewara, 2021-10-12:
please try `idf.py fullclean`

A-Babin, 2021-10-12:
> please try `idf.py fullclean`

No affect.

picli3, 2022-03-25:
> tengo el mismo problema en ubuntu20.04, idf v4.4, pero en windows funciona :(

yo lo resolvi usando el comando de crear proyectos
`idf.py create-project  NOMBRE_PROYECTO `
de esta forma te crea la estructura base, compilas y listo!

TogetDair-OCS, 2023-01-23:
d\d\