espressif/esp-idf #3609: xtensa-esp32-elf-linux64-esp32-2019r1-8.2.0.tar.gz not found on https://dl.espressif.com/dl/

no1seman, 2019-06-10:
New toolchain on page: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html is unavaible.

igrr, 2019-06-10:
Sorry, the link in the docs is incorrect. The right one is here:

https://github.com/espressif/esp-idf/blob/2c55fae6cf8de7f9776840b533acfc31e2de1b24/tools/tools.json#L27

Will fix this soon, thanks for noticing!

JustMe-NL, 2019-06-10:
Also the toolchain on https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html#toolchain-setup is missing.

The link https://dl.espressif.com/dl/xtensa-esp32-elf-osx-esp32-2019r1-8.2.0.tar.gz does not work.

igrr, 2019-06-11:
The links are generated by the same script so all 3 (windows, Mac, Linux) are incorrect. The file I have linked in the comment above does have the correct links, in the meantime. Sorry for the inconvenience caused.

closed 2019-06-12 by igrr

ryankurte, 2019-09-26:
hey there, this appears to still be broken, though it's hard to tell whether there is an issue with `dl.espressif.com` or the URL. Tested using the IDF `install.sh` script in `v4.1-dev-281-g96b96ae24` (current master) as well as the link from the [beta docs](https://docs.espressif.com/projects/esp-idf/en/feature-esp32s2beta/get-started/macos-setup.html).

AaronBottegal, 2019-11-12:
Still an issue here.

`  -> Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:01:30 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to dl.espressif.com:443 
==> ERROR: Failure while downloading https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz`

CwbhX, 2019-11-20:
Agreed it's still broken

Alvin1Zhang, 2019-11-21:
@CwbhX  @AaronBottegal @ryankurte Thanks for the updates and sorry for the slow turnaround.

strikosn, 2020-01-10:
Can you make https://dl.espressif.com/dl public?

igrr, 2020-01-13:
@strikosn Can you please clarify, are you getting some errors when installing the tools?