espressif/esp-idf #3610: ESP32 server to client

aaronchan1992, 2019-06-10:
Dear Sir,
Is it possible to implement a client to communication with multiple BLE sensor (etc ultrasonic and hall effect) for vehicle detection. I am building a BLE parking sensor for retrieving sensor data from server to client. Once the client receive it, it can display the availability data to the web server. Please guide me through. Thank you.

[ESP32 ble.zip](https://github.com/espressif/esp-idf/files/3273116/ESP32.ble.zip)

Weijian-Espressif, 2019-06-11:
@aaronchan1992,  Do you want to use multiple connections? You can refer to our gattc_multi_connect example.

aaronchan1992, 2019-06-11:
HI @Weijian-Espressif thanks for the suggestion. Based on the code, I would need to create a gatt profile for client and gatt profile for the server right?  As for my current implementation, I am matching the uuid of the server with conditions set by the client. Once it tallies it will perform BLE pairing. But beyond that how can I access the sensor data. Can you give me some advices. Thanks

closed 2019-06-18 by aaronchan1992