espressif/esp-idf #3613: How to add your own static library file using CMAKE above esp-idf

WangYifa, 2019-06-11:


WangYifa, 2019-06-11:
Can you give me a detailed procedure?

renzbagaporo, 2019-06-11:
It should be the same as in plain old CMake:

```
add_library(<library-name> STATIC <your-source-files>)
```
That should compile your source files and archive them to a <library-name>.a file. You can then link it to the enclosing component:

```
target_link_libraries(COMPONENT_LIB <PUBLIC|PRIVATE> <library-name>)
```

Or am I missing something with your question?

WangYifa, 2019-06-11:
I know how to create the .a files.
but I don't know how to add a .a file to my project. 
Can you give me an example base on the HelloWorld project？

renzbagaporo, 2019-06-11:
You mean a pre-built `.a` file, you want to link that to your project?

JasonLrh, 2019-06-12:
你可以自己写.mk规则，然后在项目Makefile中添加引用

renzbagaporo, 2019-08-12:
@WangYifa  Can this be closed?

Alvin1Zhang, 2019-08-12:
@WangYifa Thanks for reporting this, due to that [IDF issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md) has been ignored, will close the ticket for now. For questions, you can raise a post on our [forum](https://esp32.com). Thanks.

closed 2019-08-12 by Alvin1Zhang

drony, 2019-10-17:
I've been dealing with this for a long time. but I couldn't get it to work and finally decided to write here. please make a small "hello world" example.

renzbagaporo, 2019-10-18:
Ok, will do @drony.

drony, 2019-10-20:
@renzbagaporo thanks for response. i read everything about this problem. bu i can't get any result. and i'm tired:(

similiar basic sample is here. can you help me/us(i thing lot of people strugling this issue)
https://github.com/espressif/esp-idf/issues/4210

thanks is advance.