espressif/esp-idf #3614: How to close the watchdog

Herwey, 2019-06-11:
Hi All:

When I run a task, I would get the error as below:
E (13452) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13452) task_wdt:  - IDLE0 (CPU 0)
E (13452) task_wdt: Tasks currently running:
E (13452) task_wdt: CPU 0: ball_follow_tas
E (13452) task_wdt: CPU 1: IDLE1

So I want to close the watchdog, what should I do? Thanks !

Yuvaraj4041, 2019-06-11:
Hi
Goto menuconfig -> Component config ->Common ESP-related 
Uncheck 
[ ] Interrupt watchdog
[ ] Initialize Task Watchdog Timer on startup  
By default both are selected.

Dazza0, 2019-06-11:
@Herwey I would recommend against closing the task watchdog as they are automatically enabled for a good reason (i.e. it is supposed to be a safeguard against tasks that run for extended periods of time without yielding). A watchdog triggering is a symptom of an issue and not the issue itself, so I suggest you investigate what in your application code is causing the watchdog to trigger.

suda-morris, 2019-06-12:
Also you can add an entry in sdkconfig.default to disable task watchdog by default in this way : https://github.com/espressif/esp-idf/blob/master/examples/system/unit_test/test/sdkconfig.defaults#L1

Alvin1Zhang, 2019-06-18:
@Herwey Hi, would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-06-21:
@Herwey Thanks for reporting the issue. If you have further such questions, please help post on our [forum](https://esp32.com), thanks.

closed 2019-06-21 by Alvin1Zhang