espressif/esp-idf #3615: What happend when I use I2s with inner ADC and set channel_format to I2S_CHANNEL_FMT_RIGHT_LEFT

l15y, 2019-06-11:
It seems that I read 2 data at one time.
Can I  let the 2 datas from diffent adc channel?

Alvin1Zhang, 2019-06-14:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

pedroluis5, 2019-08-09:
I have the same issue, get 2 adc value equals. when I change to I2S_CHANNEL_FMT_ONLY_LEFT solve the problem, but when I increase the sample frequency the problem come back.

negativekelvin, 2019-08-09:
https://github.com/espressif/esp-idf/issues/3714#issuecomment-508995367

L-KAYA, 2024-12-18:
ADC + DMA now has dedicated driver and APIs in `adc_continuous.h` since IDF v5.0.

Close this issue since the legacy driver is no longer recommended and the IDF v4.x versions are all end of life.

closed 2024-12-18 by Alvin1Zhang