espressif/esp-idf #3616: ssl_mutual_auth_ Error : MQTT_EVENT_DISCONNECTED

Yuvaraj4041, 2019-06-11:
Hi 

I am trying the example **ssl_mutual_auth** under esp-idf.

I followed the redme file for creating certificate and key. It is compiled without any error and i flashed.

After i am checking the serial monitor and it is giving the following error.
**esp-tls: mbedtls_ssl_handshake returned -0x7280
esp-tls: Failed to open new connection
TRANS_SSL: Failed to open a new connection
MQTT_CLIENT: Error transport connect
MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED**

Here is my complete Log :
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6316
load:0x40078000,len:11304
load:0x40080400,len:6700
entry 0x40080764
[0;32mI (29) boot: ESP-IDF v4.0-dev-621-g81ca1c013-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 13:31:48[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1fba0 (129952) map[0m
[0;32mI (140) esp_image: segment 1: paddr=0x0002fbc8 vaddr=0x3ffb0000 size=0x00448 (  1096) load[0m
[0;32mI (140) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x87b10 (555792) map[0m
[0;32mI (341) esp_image: segment 3: paddr=0x000b7b30 vaddr=0x3ffb0448 size=0x02c0c ( 11276) load[0m
[0;32mI (346) esp_image: segment 4: paddr=0x000ba744 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (348) esp_image: segment 5: paddr=0x000bab4c vaddr=0x40080400 size=0x0f8e4 ( 63716) load[0m
[0;32mI (392) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (393) boot: Disabling RNG early entropy source...[0m
[0;32mI (393) cpu_start: Pro cpu up.[0m
[0;32mI (397) cpu_start: Application information:[0m
[0;32mI (401) cpu_start: Project name:     mqtt_ssl_mutual_auth[0m
[0;32mI (408) cpu_start: App version:      v4.0-dev-621-g81ca1c013-dirty[0m
[0;32mI (414) cpu_start: Compile time:     Jun 11 2019 13:31:58[0m
[0;32mI (420) cpu_start: ELF file SHA256:  37f330ba8e2ef8e0...[0m
[0;32mI (426) cpu_start: ESP-IDF:          v4.0-dev-621-g81ca1c013-dirty[0m
[0;32mI (433) cpu_start: Starting app cpu, entry point is 0x40081058[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (444) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (451) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (457) heap_init: At 3FFB9168 len 00026E98 (155 KiB): DRAM[0m
[0;32mI (463) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (469) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (476) heap_init: At 4008FCE4 len 0001031C (64 KiB): IRAM[0m
[0;32mI (482) cpu_start: Pro cpu start user code[0m
[0;32mI (164) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (166) MQTTS_EXAMPLE: [APP] Startup..[0m
[0;32mI (166) MQTTS_EXAMPLE: [APP] Free memory: 273184 bytes[0m
[0;32mI (166) MQTTS_EXAMPLE: [APP] IDF version: v4.0-dev-621-g81ca1c013-dirty[0m
I (216) wifi: wifi driver task: 3ffc0be4, prio:23, stack:3584, core=0
I (216) wifi: wifi firmware version: a13dcc4
I (216) wifi: config NVS flash: enabled
I (216) wifi: config nano formating: disabled
[0;32mI (216) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (226) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (256) wifi: Init dynamic tx buffer num: 32
I (256) wifi: Init data frame dynamic rx buffer num: 32
I (256) wifi: Init management frame dynamic rx buffer num: 32
I (266) wifi: Init management short buffer num: 32
I (266) wifi: Init static rx buffer size: 1600
I (266) wifi: Init static rx buffer num: 10
I (276) wifi: Init dynamic rx buffer num: 32
[0;32mI (276) example_connect: Connecting to Uchi_Babu...[0m
[0;32mI (366) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0[0m
I (366) wifi: mode : sta (30:ae:a4:10:91:24)
I (496) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1476) wifi: state: init -> auth (b0)
I (1476) wifi: state: auth -> assoc (0)
I (1486) wifi: state: assoc -> run (10)
I (1526) wifi: connected with Uchi_Babu, channel 1, bssid = b0:4e:26:03:37:1f
I (1526) wifi: pm start, type: 1

[0;32mI (3706) tcpip_adapter: sta ip: 192.168.0.176, mask: 255.255.255.0, gw: 192.168.0.1[0m
[0;32mI (3706) example_connect: Connected to Uchi_Babu[0m
[0;32mI (3706) example_connect: IPv4 address: 192.168.0.176[0m
[0;32mI (3716) example_connect: IPv6 address: fe80:0000:0000:0000:32ae:a4ff:fe10:9124[0m
[0;32mI (3726) MQTTS_EXAMPLE: [APP] Free memory: 236736 bytes[0m
[0;32mI (3726) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (3736) MQTTS_EXAMPLE: Other event id:7[0m
[0;31mE (4106) esp-tls: mbedtls_ssl_handshake returned -0x7280[0m
[0;31mE (4116) esp-tls: Failed to open new connection[0m
[0;31mE (4116) TRANS_SSL: Failed to open a new connection[0m
[0;31mE (4116) MQTT_CLIENT: Error transport connect[0m
[0;32mI (4116) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED[0m

closed 2019-06-12 by Yuvaraj4041

david-cermak, 2019-06-12:
Hi @Yuvaraj4041 

I noticed you've already close the issue, so I assume connection to the public broker has started working.  Let me just reference another issue https://github.com/espressif/esp-idf/issues/3120#issuecomment-472180191, where explaining about setting up a local broker and configuring for client verification -- as this should work independently on `test.eclipse.org`

Yuvaraj4041, 2019-06-12:
Hi @david-cermak 

Thanks for your suggestion. 
I am able to connect with broker. My problem is when i am downloading certificate file from mosquitto it is saving ".txt" format not ".cert" format. After solving (Converting to .cert format) this issue it is working fine.