espressif/esp-idf #3618: NVS example is not working

tatulea, 2019-06-11:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):  v3.2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description

Hi, I am trying to use the nvs_rw_value example and it doesn't work at all. It tells me all the time that there are no values in the flash, after each reset.
I replaced the reset line with `while(1)` and I can see that my reset number increments from flash sometimes when I restart the board by cutting off the power supply. I couldn't find an exact pattern when it works and when it doesn't.

Any idea about the problem?

tatulea, 2019-06-12:
I have an I2C device connected to my module and SCL is on GPIO14. It looks like this is the problem. If I disconnect the SCL, power down the device and power it up again, the NVS works. If I reconnect the SCL during the running of the program, it will work as well. 

So it doesn't work when I have the SCL connected and power up the board (not restarting).

I think it may be related to the fact that GPIO14 is MTMS, which is a pin for JTAG. I don't need the JTAG in my product. How can I solve this?

igrr, 2019-06-12:
I doubt the issue could be caused by GPIO14, however do you also use GPIO12 for any purpose?

tatulea, 2019-06-12:
Yes, it is the interrupt from the IC that uses the GPIO12 as SCL.

igrr, 2019-06-12:
In that case, please keep in mind that GPIO12 is a strapping pin which determines VDD_SDIO (flash) voltage.

If this line is high when ESP32 is reset (by power-on or by the RTC watchdog), then VDD_SDIO will be 1.8V. If low, VDD_SDIO will be 3.3V. Since you are using ESP32-WROVER-B, flash voltage should be 3.3V, and GPIO12 should be kept low on reset. See section 2.3 "Strapping pins" of the [WROVER-B datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-b_datasheet_en.pdf) for more details.

The cause of the issue you observe is likely that GPIO12 is high on reset, setting VDD_SDIO voltage to 1.8V. The flash chip (rated for 3.3V) can still work for reading, most of the time, but fails to write or erase data. To fix the issue, you need to ensure that GPIO12 is kept low low on reset.

You can force also VDD_SDIO to be 3.3V, regardless of the GPIO12 level at reset, by burning EFUSEs, see [Setting Flash Voltage](https://github.com/espressif/esptool/wiki/espefuse#setting-flash-voltage-vdd_sdio) for instructions.

tatulea, 2019-06-12:
Thanks, I will try this tomorrow.
Anyway, why is it working when I am disconnecting the SCL?

igrr, 2019-06-12:
I can't say for sure, but try measuring the voltage on GPIO12 in this case.

tatulea, 2019-06-13:
The voltage on GPIO12 is 0.9V, but I burnt the efuse with flash 3.3V and it is working now. Thanks for the help!

closed 2019-06-13 by tatulea