espressif/esp-idf #3621: Nimble component Kconfig has old BTDM_CONTROLLER_HCI_MODE_VHCI

mwneuendorf, 2019-06-12:
Line 3 of components/nimble/Kconfig.in
should be BTDM_CTRL_HCI_MODE_VHCI.

dhrishi, 2019-06-12:
@mwneuendorf  The NimBLE Kconfig has old names for the config options because it is based on an older version of ESP-IDF. It should work fine. Did you see any issue related to compilation?

This will be changed when it gets merged in master branch.

mwneuendorf, 2019-06-12:
yes when running make menuconfig. nimble does not show up in the component bluetooth section when doing a clone and checkout of the latest code base.

dhrishi, 2019-06-12:
Can you please check if your IDF_PATH is correct?

I am able to see the NimBLE component in menuconfig and also compilation of `bleprph` app works fine on latest `feature/nimble-preview` branch

mwneuendorf, 2019-06-12:
ok. I tried to pull the recent code base with
git pull . origin/feature/nimble-preview

that didn't work. 
From .
 * remote-tracking branch origin/feature/nimble-preview -> FETCH_HEAD
Auto-merging docs/en/api-reference/bluetooth/index.rst
CONFLICT (content): Merge conflict in docs/en/api-reference/bluetooth/index.rst
Auto-merging docs/Doxyfile
Auto-merging components/esp32/ld/esp32.project.ld.in
Auto-merging components/bt/include/esp_bt.h
Auto-merging components/bt/bt.c
Auto-merging components/bt/Kconfig
Adding as components/bootloader/subproject/components/micro-ecc/micro-ecc~HEAD instead
Auto-merging .gitmodules
CONFLICT (content): Merge conflict in .gitmodules
CONFLICT (modify/delete): .gitlab-ci.yml deleted in d8e4ec91e04f8f07366ae97d88ea1cfb1cda88a8 and modified in HEAD. Version HEAD of .gitlab-ci.yml left in tree.
Automatic merge failed; fix conflicts and then commit the result.

this seems to be safe but it doesn't pull the most recent code.
git clone https://github.com/espressif/esp-idf --single-branch --branch feature/nimble-preview
git submodule update --init --recursive

is there a safer way of updating to the recent code? or is that not possible right now?

dhrishi, 2019-06-12:
Ideally, git pull should work fine. In case of the error you see above, you can just reset to the current head of the branch `git reset --hard origin/feature/nimble-preview`

mwneuendorf, 2019-06-12:
ok. that seemed to do the trick.

closed 2019-06-12 by mwneuendorf