espressif/esp-idf #3622: nvs_partition_generator.py fails to read input csv file

michelesolytic, 2019-06-12:
## Environment

- Development Kit:  Custom board
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):  v3.1.4-2-gf83305ab2
- Operating System: Fedora 29
- Python version: Python 3.7.3

## Problem Description

After upgrading ESP-IDF from version 3.1.3 to 3.1.4, nvs_partition_generator.py crashes while attempting to create a partition binary with the following command:
`python3 esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --version v1 nvs_local.csv nvs.bin`

The traceback is given as follows:

```
Traceback (most recent call last):
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 726, in <module>
    main()
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 723, in main
    nvs_part_gen(input_filename, output_filename, key_gen, is_encrypt_data, key_file, version_no)
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 649, in nvs_part_gen
    for row in reader:
  File "/usr/lib64/python3.7/csv.py", line 111, in __next__
    self.fieldnames
  File "/usr/lib64/python3.7/csv.py", line 98, in fieldnames
    self._fieldnames = next(self.reader)
_csv.Error: iterator should return strings, not bytes (did you open the file in text mode?)
```

Replacing line 645 with the following fixes the issue (basically just remove the 'b'):
`input_file = open(input_filename, 'r')`

michelesolytic, 2019-06-12:
Further testing shows that it doesn't work anymore with string fields though, such as:
`TimePath,data,string,/api/sensor/tzo`
Traceback:
```
Traceback (most recent call last):
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 726, in <module>
    main()
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 723, in main
    nvs_part_gen(input_filename, output_filename, key_gen, is_encrypt_data, key_file, version_no)
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 651, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 581, in write_entry
    nvs_instance.write_entry(key, value, encoding)
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 513, in write_entry
    self.cur_page.write_varlen_data(key, value, encoding, self.namespace_idx,self)
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 392, in write_varlen_data
    self.write_single_page_entry(entry_struct, data, datalen, data_entry_count, nvs_obj)
  File "/home/michele/git/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py", line 325, in write_single_page_entry
    crc = zlib.crc32(data, 0xFFFFFFFF)
TypeError: a bytes-like object is required, not 'str'
```

dobairoland, 2019-06-12:
Hi @michelesolytic. Please note that Python3 is not supported in v3.1. Please continue to use Python2.

If it fails for `import builtins` then please try to install the `python2-future` package from the Fedora repository.

dobairoland, 2019-06-27:
Hi @michelesolytic. Does your issue get solved?

dobairoland, 2019-07-03:
I'm closing this for lack of activity.

closed 2019-07-03 by dobairoland