espressif/esp-idf #3623: NimBLE: ble_svc_gap.c fails to compile when build configured to detect uninitialized variable

bfriedkin, 2019-06-12:
Hello -

I get the following error when my build is configured to generate errors on uninitialized variables:

```
/Users/brianfriedkin/esp32/esp-idf/components/nimble/nimble/nimble/host/services/gap/src/ble_svc_gap.c: In function 'ble_svc_gap_access':
/Users/brianfriedkin/esp32/esp-idf/components/nimble/nimble/nimble/host/services/gap/src/ble_svc_gap.c:195:9: error: 'rc' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     int rc;
```

Initializing `rc` to `0` resolves the problem, though I don't know if that's a preferred solution.

Regards,
Brian

dhrishi, 2019-06-13:
@bfriedkin Thanks. I will fix this

dhrishi, 2019-07-23:
This has been fixed with commit 416e753b7c3e539c44e0fb743f22c75bff7a42b6

closed 2019-07-26 by mahavirj