espressif/esp-idf #3625: "xTaskCreatePinnedToCore" didn't work

Herwey, 2019-06-13:
Hi All:
I used this "xTaskCreatePinnedToCore(ball_follow_task, "ball_follow_task", 1024*10, NULL, 10, &xTask5,tskNO_AFFINITY) “ to create a task, but it didn't work ?! At the same time , I had other twos task running:
    xTaskCreatePinnedToCore(i2c_task, "i2c_task", 1024 * 2, (void *)0, 8, NULL,tskNO_AFFINITY);
    xTaskCreatePinnedToCore(send_IIC, "send_IIC", 1024*10, NULL, 8, NULL,tskNO_AFFINITY);
So, why the ball_follow_task didn't . Actually, yesterday, they worked fine.
Thanks in advance !

chegewara, 2019-06-13:
```
err = xTaskCreatePinnedToCore(ball_follow_task, "ball_follow_task", 1024*10, NULL, 10, &xTask5,tskNO_AFFINITY);
if(err != ESP_OK)
probably not enough memory
```

Herwey, 2019-06-14:
Hi
I used as below:
    err=xTaskCreatePinnedToCore(ball_follow_task, "ball_follow_task", 1024*10, NULL, 10, &xTask5,tskNO_AFFINITY);
     if(err != ESP_OK)
    printf("probably not enough memory");

It printed "probably not enough memory", is not "ESP_OK”， SO what should I do next ? Thx!

chegewara, 2019-06-14:
Check code and see why you are short on heap.

Maybe you have typo in your code here and this task is taking all heap:
`xTaskCreatePinnedToCore(send_IIC, "send_IIC", 102410`

Herwey, 2019-06-14:
That just a type error. Now I only run one task as below:
err=xTaskCreatePinnedToCore(ball_follow_task, "ball_follow_task", 1024*25, NULL, 10, &xTask5,tskNO_AFFINITY);
     if(err != ESP_OK)
     {printf("probably not enough memory");}
     else{
        printf("ok" );
     }

But it showed "not enough memory".....

projectgus, 2019-06-14:
@Herwey FreeRTOS APIs like xTaskCreatePinnedToCore() don't return `esp_err_t`, they return the FreeRTOS constant pdPASS on success which is defined to "pdTRUE==1.

ESP_OK enum is not used by FreeRTOS APIs but has integer value 0.

So probably the function is succeeding. Print the actual result integer to confirm, or check if the task is actually running by printing something from the task function.

Herwey, 2019-06-14:
Thanks ! As below:
    int result=0;
    result = xTaskCreatePinnedToCore(ball_follow_task, "ball_follow_task", 1024*25, NULL, 10, &xTask5,tskNO_AFFINITY);
     printf("The result is %d \r\n", result);
The "result" is 1. I think the createtask success. There is a "while(true)" loop in my  task, but it just run some loops. Then the program gets stuck. Just like a pause

projectgus, 2019-06-14:
@Herwey Yes, your task is being created.

FreeRTOS is a real-time operating system so if you have a task running a `while(1)` loop then it will prevent any lower priority task from running on that CPU. There are two CPUs in an ESP32 but if two tasks are doing this, nothing else will run.

If you want a particular task to stop running for a while, try calling vTaskDelay(). Even putting vTaskDelay(1) in the `while(1)` loop will probably solve this problem.

Herwey, 2019-06-14:
But when I only run one task , it also will pause sometimes , I also add "vtaskdelay(10)" in the loop.
xTaskCreatePinnedToCore(ball_follow_task, "ball_follow_task", 1024*25, NULL, 10, &xTask5,tskNO_AFFINITY);

Herwey, 2019-06-14:
Do you think it about the driver in "esp-idf" ?

Herwey, 2019-06-14:
I close the Watchdog, is it ok?

Herwey, 2019-06-14:
E (14695) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (14695) task_wdt:  - IDLE0 (CPU 0)
E (14695) task_wdt: Tasks currently running:
E (14695) task_wdt: CPU 0: ball_follow_tas
E (14695) task_wdt: CPU 1: IDLE1
E (19695) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (19695) task_wdt:  - IDLE0 (CPU 0)
E (19695) task_wdt: Tasks currently running:
E (19695) task_wdt: CPU 0: ball_follow_tas
E (19695) task_wdt: CPU 1: IDLE1

I openned the watchdog, now it showed me the error like this. Thanks !

projectgus, 2019-06-14:
@Herwey the task watchdog is saying that `ball_follow_tas` on CPU0 is not delaying or blocking often enough to allow other tasks to run.

Without seeing your code it's not possible to guess what any other problems might be.

This is probably not a bug in ESP-IDF, but something with the code. If you can't find the solution, try posting as much code as possible (or the whole project) on https://esp32.com and someone will probably be able to help you there.

closed 2019-06-14 by projectgus