espressif/esp-idf #3628: [esp_http_client] Incorrect Host header for non-default ports (missing port)

viettrungluu, 2019-06-14:
IDF version: v3.2 (but also on v3.3 and current master)

## Problem Description

When connecting to a non-default port (80 for HTTP, 443 for HTTPS), the Host header sent is incorrect; in those cases, it should include the port. It would be valid to always include the port, though.

Code (from current v3.2) here: https://github.com/espressif/esp-idf/blob/release/v3.2/components/esp_http_client/esp_http_client.c#L527 (note that the Host header is set in several different places).

RFC2616 14.23 here: https://www.w3.org/Protocols/rfc2616/rfc2616-sec14.html. It says:

> The Host request-header field specifies the Internet host and port number of the resource being requested, as obtained from the original URI given by the user or referring resource (generally an HTTP URL, as described in section 3.2.2). The Host field value MUST represent the naming authority of the origin server or gateway given by the original URL.

and

> A "host" without any trailing port information implies the default port for the service requested (e.g., "80" for an HTTP URL).

(Ironically, this is mainly a problem when running against strict test servers. Of course, failing to run further tests because you fail right at the start is a bigger problem.)

### Expected Behavior

If I perform a request to "http://www.example.com:8080", the header should be `Host: www.example.com:8080`.

### Actual Behavior

The header is actually `Host: www.example.com` (which implies port 80).

jitin17, 2019-06-17:
@viettrungluu Thanks for bringing this to our notice. We will soon fix this.

viettrungluu, 2019-06-17:
@jitin17 Thanks. FWIW, I have a patch (on top of a slightly-modified v3.2) here: https://github.com/tricot-io/esp-idf/commit/c9d82fb2fd6fca2ba0db8bbbb825dfab04a9c033

(But obviously this patch needs substantial changes for v3.3/v4.0/master.)

jitin17, 2019-06-18:
@viettrungluu This looks good tricot-io@c9d82fb. Thanks!! 
If you wish to raise a PR with this fix, you can do it. Otherwise, I'll try to add this fix.

viettrungluu, 2019-06-18:
@jitin17 Sure, I can make a pull request. I guess you want a pull request to master?

viettrungluu, 2019-06-18:
@jitin17 FYI, I noticed that v3.2 (the tag), `esp_http_client_set_url()` has the bug that `old_host` becomes invalid after `http_utils_assign_string(&client->connection_info.host()`. The HEAD of `release/v3.2` fixes this (by `strdup()`ing), but in turn it leaks `old_host` in a number of failure cases (the `HTTP_MEM_CHECK`s before `old_host` is freed). So I'll clean that up too.

viettrungluu, 2019-06-18:
@jitin17 I opened a pull request on the release/v3.2 branch. Unfortunately, I haven't been able to test it, since for some reason WiFi won't connect on release/v3.2 HEAD (nor on master, as of a few days ago); I haven't investigated this issue yet -- do you happen to know anything about it?

jitin17, 2019-06-19:
@viettrungluu Thanks for raising the PR. I'll try to test and integrate it.

closed 2020-10-14 by espressif-bot