espressif/esp-idf #3629: CONFIG_FREERTOS_PORTMUX_DEBUG causes crash on nvs_flash_init()

william-ferguson-au, 2019-06-14:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v3.3-beta3-34-g8c57aa024
- Build System:         CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         Battery

## Problem Description

//Detailed problem description goes here.
Invoking nvs_flash_init() with CONFIG_FREERTOS_PORTMUX_DEBUG configured causes crash.

### Expected Behavior
No crash, and I would get some debug about the FREERTOS_PORTMUX which seems to have a fault that I am trying to track down or avoid.

### Actual Behavior
Crash on invocation of nvs_flash_init()

### Steps to reproduce

1. compile and flash.
2. On boot note crash.

### Code to reproduce this issue

```cpp
#include <esp_log.h>
#include <nvs_flash.h>

static void app_main() {
    Serial.begin(115200);
    ESP_LOGI(TAG, "Logger initialised");

    // Initialize NVS.
    esp_err_t err = nvs_flash_init();

    // Never get here
    ESP_LOGI(TAG, "Finished setup");
}
```

## Debug Logs

```
--- idf_monitor on COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6332
load:0x40078000,len:11308
ho 0 tail 12 room 4
load:0x40080400,len:6700
entry 0x40080764
I (30) boot: ESP-IDF v3.3-beta1-728-g8c57aa024 2nd stage bootloader
I (30) boot: compile time 17:58:47
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00005000
[... 126 lines cut ...]

0x400d7076: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:865 (discriminator 3)

0x400d2ec5: fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50

0x400d2e54: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)

0x40086fb2: vPortCPUReleaseMutexIntsDisabledInternal at E:/Dev/Espressif/esp-idf/components/freertos/tasks.c:4590
 (inlined by) vPortCPUReleaseMutexIntsDisabled at E:/Dev/Espressif/esp-idf/components/freertos/portmux_impl.h:109
 (inlined by) vTaskExitCritical at E:/Dev/Espressif/esp-idf/components/freertos/tasks.c:4289

0x40087978: xTaskResumeAll at E:/Dev/Espressif/esp-idf/components/freertos/tasks.c:4590

0x40083116: spi_flash_op_block_func at E:/Dev/Espressif/esp-idf/components/spi_flash/cache_utils.c:201

0x4008201b: ipc_task at E:/Dev/Espressif/esp-idf/components/esp32/ipc.c:62

0x40085831: vPortTaskWrapper at E:/Dev/Espressif/esp-idf/components/freertos/port.c:389

Rebooting...
```

## Other items if possible

- [ ] sdkconfig file (attached)

william-ferguson-au, 2019-06-14:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3291622/sdkconfig.txt)

mahavirj, 2019-06-21:
@william-ferguson-au Can you please try increasing following timeout (maybe by factor of 10)?

https://github.com/espressif/esp-idf/blob/58df1d93bc17c74499d58e05390af9c309192a5c/components/freertos/portmux_impl.inc.h#L54

william-ferguson-au, 2019-06-24:
Increasing to 100,000 (10 fold increase) - same behaviour
Increasing to 1,000,000 (100 fold increase) - same behaviour

mahavirj, 2019-06-24:
@william-ferguson-au I do not see any other change (with debug macros enabled) that can cause this issue. Sorry to trouble you again but can you please try to comment out complete `if` block from:

https://github.com/espressif/esp-idf/blob/58df1d93bc17c74499d58e05390af9c309192a5c/components/freertos/portmux_impl.inc.h#L53

william-ferguson-au, 2019-06-24:
Commenting out that entire `if` block seems to work.

Alvin1Zhang, 2020-05-29:
Thanks for reporting and sorry for slow turnaround. The fix has been available at https://github.com/espressif/esp-idf/commit/e787cb1730fca6514fbceeb7d26d5688e829f77f. Feel  free to reopen if the issue still happens. Thanks.

closed 2020-05-29 by Alvin1Zhang

diplfranzhoepfinger, 2020-08-02:
Hello, 
would be nice if this would be fixed also in 4.1 and 4.2. THANKS !

projectgus, 2020-08-04:
Hi @diplfranzhoepfinger ,

The change that fixes this issue should already be part of the v4.2 release branch. Are you still seeing this issue in v4.2?

We'll see about fixing this bug in v4.1 as well.

diplfranzhoepfinger, 2020-08-04:
i did not test in 4.2

4.1 yes has also this issue..

projectgus, 2020-08-04:
I should clarify that the "fix" Alvin and I mentioned is actually that portMUX debugging was removed in v4.2 and newer, when we refactored to use a non-FreeRTOS-specific "spinlock" function (the normal portMUX macros continue to work as wrappers around spinlocks).

We removed portMUX debugging because we found that generally it wasn't useful - usually if there's a bug with portMUX then it's due to a function getting stuck in a critical section, and the [interrupt watchdog](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#interrupt-watchdog) can find this more effectively. Most other portMUX-related bugs are related to timing, and adding debugging to every portMUX invocation changes the timing completely - so you end up debugging different behaviour to what you were originally trying to debug.

Although we try not to ever make these kind of breaking changes, we decided it was better to remove an ineffective debugging tool than to keep it in the codebase.

@diplfranzhoepfinger Is there something in particular that you're trying to debug, that we can help with?

Angus

diplfranzhoepfinger, 2020-08-05:
ah, thanks so far ...

I am debugging several things. i will come back to you later.