espressif/esp-idf #3630: idf_build_component() broken since 065eb918f

PerMalmberg, 2019-06-14:
The new CMake changes merged into master in 065eb918f broke idf_build_component(), at least when used from projects that uses IDF as a library - my sdkconfig is not longer getting configuration values from my `Kconfig` in my configuration-only component, nor does menuconfig display the options. It works fine in 8886e3fb.

I see nothing in the latest docs that the use of this function has changed, only that it must be called before idf_build_process, which I do.

I don't have a small example available, but you can see the issue by compiling https://github.com/PerMalmberg/Smooth using the two above mentioned commits of IDF and xtensa-gcc-8. There are some shell scripts available in the root folder of the project that shows how I build it.

renzbagaporo, 2019-06-15:
Thanks again for the small example. The only thing I'm noticing in it is that no other component seem to require `extra_component`, either publicly or privately. 

This was a bug in the previous implementation. The comment:

```
                    # try and trim the build; additional components
                    # will be included as needed based on dependency tree
```

was obviously not how it happens pre-`065eb91`.

To solve this either have another component add `extra_component` to its requirements (that component should also be added by another component as part of a dependency chain, so on and so forth) or also add `extra_components` to `COMPONENTS` argument to `idf_build_process`.

Agree that docs could be clearer about this though.

PerMalmberg, 2019-06-15:
Ok, that makes total sense.

Perhaps I'm reading too much into the name "idf_build_component"; to me it means it'll be built regardless. A name like "idf_process_component" would make it more clear what it actually does. Change it or leave it as you see fit, I consider this issue closed.

Thank you @renzbagaporo

renzbagaporo, 2019-06-15:
The `build` is used as a noun, `idf_build_component` as in "this is a component in the build" (which may or may not be used depending on the dependency tree).

Reopening the issue, as clarification of docs is the better fix ;)

closed 2019-06-26 by igrr