espressif/esp-idf #3634: The "qr_task" can't be created successful when the wifi is openning(AP)

Herwey, 2019-06-15:
Hi All:
I open the wifi(AP) as below:
 wifi_init_softap();  
 vTaskDelay(100 / portTICK_PERIOD_MS); 
 http_server_init();  
Then I created the qr_reco task as below:
xTaskCreate( qr_recognize, "qr_recognize", 1024*60, &camera_config, 8, &xTask4);
But the task responsed "-1", maybe not enough memory. The depth is "1024*60", but when I created a task which depth is "1024*20", it could work well. Does the wifi cost lots of memory? So what can I do next ? Thanks ! :D

negativekelvin, 2019-06-15:
Do not use such a huge stack, use heap memory with malloc and use wrover with external ram if necessary.

Herwey, 2019-06-15:
How to use the external ram ? Because the QR task needs the huge stack, if smaller, it will overflow. Use the external ram maybe the only way. but I can't find it in "make menuconfig". How ? 
Thanks :P

negativekelvin, 2019-06-15:
Are you using this https://github.com/m5stack/m5stack-cam-psram

Herwey, 2019-06-15:
Yes, so the task is very huge. When I open the Wifi. I can't create the qr task.

negativekelvin, 2019-06-15:
They say it should work with psram, so you should open an issue on their repo if it is not working

igrr, 2019-06-15:
@Herwey if you are using quirc library, you may need to look into the image segmentation function which it uses. Replace the recursive segmentation function with a queue-based one, and the stack size requirements will become much more reasonable.

In any case, yes, the issue is that the task creation fails because there isn't a free continuous region of RAM large enough to place the stack.

Herwey, 2019-06-16:
Emmm, how to replace the Replace the recursive segmentation function... I'm a new beginner :D

projectgus, 2020-10-11:
Closing as this is an (old) issue with m5stack-cam-psram project, rather than an issue with ESP-IDF itself.

If further assistance with this project is needed, suggest either opening an issue on their project or starting a discussion on the https://esp32.com forum.

closed 2020-10-11 by projectgus