espressif/esp-idf #3635: [question] heap_caps_print_heap_info

chegewara, 2019-06-15:
Hi, 
i like numbers so if i cant explain something it bothers me.
In AWSFreeRTOS release-1.5, which is based on esp-idf 3.1.3, i have this code:
```
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    esp_bluedroid_disable();
    esp_bluedroid_deinit();
    esp_bt_controller_disable();
    esp_bt_controller_deinit();
    ESP_ERROR_CHECK( esp_bt_controller_mem_release( ESP_BT_MODE_BLE ) );
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
```
and logs output something like that:
```
Heap summary for capabilities 0x00001000:
I (2515) wifi: state: auth -> assoc (0)
  At 0x3ffbbb28 len 8192 free 0 allocated 8008 min_free 0
    largest_free_block 0 alloc_blocks 38 free_blocks 0 total_blocks 38
I (2515) wifi: state: assoc -> run (10)
  At 0x3ffb2730 len 22736 free 0 allocated 22392 min_free 0
    largest_free_block 0 alloc_blocks 78 free_blocks 0 total_blocks 78
  At 0x3ffaff10 len 240 free 0 allocated 196 min_free 0
    largest_free_block 0 alloc_blocks 3 free_blocks 0 total_blocks 3
  At 0x3ffce180 len 73344 free 16 allocated 72520 min_free 16
    largest_free_block 4 alloc_blocks 190 free_blocks 4 total_blocks 194
  At 0x3ffe0440 len 15296 free 96 allocated 15092 min_free 24
    largest_free_block 80 alloc_blocks 17 free_blocks 2 total_blocks 19
  At 0x3ffe4350 len 113840 free 83488 allocated 30268 min_free 74880
    largest_free_block 83488 alloc_blocks 12 free_blocks 1 total_blocks 13
  Totals:
    free 83304 allocated 148764 min_free 74920 largest_free_block 83160

Heap summary for capabilities 0x00001000:
  At 0x3ffbdb28 len 9432 free 9396 allocated 0 min_free 9396
    largest_free_block 9396 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffb8000 len 15144 free 15068 allocated 36 min_free 15068
    largest_free_block 15068 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffb0000 len 10032 free 9956 allocated 36 min_free 9956
    largest_free_block 9956 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffae6e0 len 6192 free 6116 allocated 36 min_free 6116
    largest_free_block 6116 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffbbb28 len 8192 free 7832 allocated 284 min_free 0
    largest_free_block 6628 alloc_blocks 9 free_blocks 2 total_blocks 11
  At 0x3ffb2730 len 22736 free 22660 allocated 36 min_free 0
    largest_free_block 22660 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffaff10 len 240 free 0 allocated 196 min_free 0
    largest_free_block 0 alloc_blocks 3 free_blocks 0 total_blocks 3
  At 0x3ffce180 len 73344 free 16096 allocated 56668 min_free 16
    largest_free_block 13388 alloc_blocks 126 free_blocks 11 total_blocks 137
  At 0x3ffe0440 len 15296 free 128 allocated 15060 min_free 0
    largest_free_block 80 alloc_blocks 16 free_blocks 3 total_blocks 19
  At 0x3ffe4350 len 113840 free 83160 allocated 30588 min_free 74880
    largest_free_block 83160 alloc_blocks 14 free_blocks 1 total_blocks 15
  Totals:
    free 170412 allocated 102940 min_free 113736 largest_free_block 83160
```
What bothers me is difference between `free+allocated`:
` free 83304 allocated 148764 min_free 74920 largest_free_block 83160` = ~230kB
vs
`free 170412 allocated 102940 min_free 113736 largest_free_block 83160` = ~270kB

negativekelvin, 2019-06-15:
Why does that bother you? When the bt memory is released it is added to heap.

chegewara, 2019-06-15:
Because if there is no consistency then i cant rely on values im getting from that function.
If releasing memory from BLE is causing that `allocated` memory is 100kB and before that it was 148kB, that means i should have about 48kB more memory free. But from logs i see i have almost 90kB more memory free. 
Do you see difference? Its not 1 or 2 kB, its about 40kB difference.

negativekelvin, 2019-06-15:
Change to allocated memory is not coming from mem release because that is reserved memory, total heap size is expected to change.  It could come from shutting down bluedroid.

mahavirj, 2019-08-28:
@chegewara Can this issue be closed?

closed 2019-08-29 by chegewara