espressif/esp-idf #3636: esp32.project.ld

PerMalmberg, 2019-06-15:
This isn't a bug report, but rather a question on the correct way to do things.

I I use IDF as a library so my dependency tree is as follows:

main app -> library -> IDF

I've not built the main app for a while so I'm not sure when this broke, but with the latest IDF (v4.0-dev-837-g58df1d93b)I'm getting this linker error when linking the final binary:

> /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot open linker script file esp32.project.ld: No such file or directory

So, whta's the correct way to set up the include directory for this linker script?

PerMalmberg, 2019-06-16:
Closing this as I'm trying to switch to compiling my lib as an IDF-component thus making this question obsolete.

renzbagaporo, 2019-06-17:
Unclosing as the dependency you mentioned should work (main app -> library -> IDF). This was one of the goals with the changes in our CMake build internals. So in the case of the `idf_as_lib` example, it would be linking like as follows:

```
    add_library(my_interface INTERFACE) # some intermediate library
    target_link_libraries(my_interface INTERFACE idf::esp32 idf::freertos idf::spi_flash)
    target_link_libraries(${elf_file} my_interface)
    idf_build_executable(${elf_file})
```

instead of linking IDF libraries to the executable directly, correct?

PerMalmberg, 2019-06-17:
Yes, looks pretty much to what I tried.

renzbagaporo, 2019-06-19:
So I tried building your project, https://github.com/PerMalmberg/Smooth, I assume? Like I said, this is supposed to be a supported use case so I'm treating this as a bug. Doesn't seem to end up in the error you mentioned. Anything special I have to do to reproduce?

PerMalmberg, 2019-06-19:
Hm, that's unexpected. I'm quite sure I got that error when I compiled the master branch at the time of the original post. Perhaps I had some leftover artifacts that caused the issue? :confused:

renzbagaporo, 2019-06-20:
I'm not aware of leftover artifacts causing this kind of issue. FWIW, here are the steps I performed though:

1. Clone https://github.com/PerMalmberg/Smooth
2. Checkout 58df1d93 on ESP-IDF repo
3. Run ./build.sh on smooth project dir

PerMalmberg, 2019-06-22:
@renzbagaporo I stubled upon this again, I had accidentally commented out a call to `idf_build_executable`. Restoring the call solves it.

renzbagaporo, 2019-06-24:
That's great. I will be closing this then @PerMalmberg.

closed 2019-06-24 by renzbagaporo