espressif/esp-idf #3637: Lack of CMakeLists.txt in component creates extremely confusing build errors.

william-ferguson-au, 2019-06-16:
## Environment

- Development Kit:      NA
- Module or chip used:  NA
- IDF version : v3.3-beta3-34-g8c57aa024
- Build System:         CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:   Windows
- Power Supply:         NA

## Problem Description

If a component is listed as a dependency, but is missing its CMakeLists.txt file then the error message from the build points you in entirely the wrong direction.
Ie libA lists libB as  a dependency but libB is missing CMakeLists.txt

### Expected Behavior

That the build error message points you towards the problem.
Ie that it says there is a problem because libB is referenced as a dependency but cannot be found on the component path.

### Actual Behavior

It says that libA " is already used to build a source directory.  It cannot be used to build
  source directory" pathToLibA. "Specify a unique binary directory name."

```
CMake Error at E:/Dev/Espressif/esp-idf/CMakeLists.txt:121 (add_subdirectory):
  The binary directory

    E:/Source/esp32/wt_trinity/build/esp-idf/libA

  is already used to build a source directory.  It cannot be used to build
  source directory

    E:/Source/esp32/wt_trinity/components/libA

  Specify a unique binary directory name.

-- Building empty aws_iot component due to configuration
-- Component libraries:
-- Configuring incomplete, errors occurred!
See also "E:/Source/esp32/wt_trinity/build/CMakeFiles/CMakeOutput.log".
See also "E:/Source/esp32/wt_trinity/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja
E:\Dev\CMake\bin\cmake.exe -SE:\Source\esp32\wt_trinity -BE:\Source\esp32\wt_trinity\build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
```

### Steps to repropduce

1. components/libA/CMakeLists.txt has:
```
SET (COMPONENT_PRIV_REQUIRES libB)
 ```
2. components/libB/CMakeLists.txt - does NOT exist.
3. idf.py build

william-ferguson-au, 2019-06-16:
This all stems from #find_component_path in  component_utils.cmake

If as indicated in the method comment it through a FATAL_ERROR if the component is not found we would have a really good error message. Ie

```
# Given a component name (find_name) and a list of component paths (component_paths),
# return the path to the component in 'variable'
#
# Fatal error is printed if the component is not found.
function(find_component_path find_name components component_paths variable)
    list(FIND components ${find_name} idx)
    if(NOT idx EQUAL -1)
        list(GET component_paths ${idx} path)
        set("${variable}" "${path}" PARENT_SCOPE)
        return()
    else()
    endif()
    # TODO: find a way to print the dependency chain that lead to this not-found component
    message(FATAL_ERROR "Required component ${find_name} is not found in any of the provided COMPONENT_DIRS")
endfunction()
```

renzbagaporo, 2019-06-18:
Thanks for reporting this issue. We'll be taking a look at this as soon as possible.

renzbagaporo, 2019-08-12:
This should be fixed @william-ferguson-au

closed 2019-08-12 by renzbagaporo