espressif/esp-idf #3638: ESP flashing through on board micro controller

hvishu, 2019-06-16:
Hi,
In one of our models, we have Renesas microcontroller along with ESP32 controller.  Renesas is kind of Maser.  Renesas communicates with ESP32 through UART (UART1, IO4 and IO15).  We would like to understand the possibility of flashing the ESP32 through Renesas, is this something possible?  Any example projects/code that I can refer for it?

Regards,
Vishwa

negativekelvin, 2019-06-16:
https://github.com/espressif/esp-idf/tree/master/examples/system/ota/native_ota_example

Instead of getting the data from http, get it from uart.  But does reneasas have the ability to deliver the large app binary?  Using USB?

hvishu, 2019-06-17:
Not using USB, using UART of Renesas.   In production line, the UART connector of Renesas is accessible to PC, so we are hoping that use Renesas UART to program ESP flash.
So, with the above examples, we would be able to flash the ESP fully (including bootloader, partitions, main binary)?

negativekelvin, 2019-06-17:
No, that assumes it already is running a firmware that supports ota. You have to do the first flash through uart0 default pins.

hvishu, 2019-06-17:
Ok.  Understood

closed 2019-06-17 by hvishu

igrr, 2020-01-29:
We have published a library for flashing ESP32 from another MCU: https://github.com/espressif/esp-serial-flasher.
The only host MCU supported now is ESP32, but the library should be relatively easy to port.