espressif/esp-idf #3642: [BLE Mesh] When client/server model publish long data, a lot of warnings will be prompted

zhanlej, 2019-06-17:
## Problem Description

Hi, 
When I want to test the maximum data that ble mesh calls ``esp_ble_mesh_client_model_send_msg`` and ``esp_ble_mesh_model_publish`` can send, I find that when the data length exceeds 8 bytes, the devices at both ends will prompt a lot of warnings. Is that correct?

### Code to reproduce this issue

client model :
```cpp
void size_test_task(void *pvParameter)
{
    int i = 0, j = 0, err = 0;
    unsigned char send_buf[500] = {0};

    for(i = 1; i < 500; i++)
    {
        for(j = 0; j < i; j++)
        {
            send_buf[j] = j;
        }

        ESP_LOGI(TAG, "%s: client publish i:%d", __func__, i);
        err = esp_ble_mesh_model_publish(iot_ble_mesh_client.model, IOT_BLE_MESH_VND_MODEL_OP_SET,
                                            i, send_buf, ROLE_NODE);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "%s: Publish iot ble mesh server Status failed", __func__);
        }

        vTaskDelay(30000 / portTICK_PERIOD_MS);
    }

    while(1);
}
```

server model :
```cpp
        case IOT_BLE_MESH_VND_MODEL_OP_SET:
            ESP_LOGI(TAG, "%s: IOT BLE MESH server receives msg, opcode 0x%04x", __func__, param->model_operation.opcode);
            struct net_buf_simple buf = {
                .len = param->model_operation.length,
                .data = param->model_operation.msg,
            };
            ESP_LOG_BUFFER_HEX("iot ble mesh server recv", buf.data, buf.len);
            break;
```

## Debug Logs

client model :
```
I (42992) ble_mesh_client: size_test_task: client publish i:1
I (73002) ble_mesh_client: size_test_task: client publish i:2
I (103002) ble_mesh_client: size_test_task: client publish i:3
I (133002) ble_mesh_client: size_test_task: client publish i:4
I (163002) ble_mesh_client: size_test_task: client publish i:5
I (193002) ble_mesh_client: size_test_task: client publish i:6
I (223002) ble_mesh_client: size_test_task: client publish i:7
I (253002) ble_mesh_client: size_test_task: client publish i:8
I (283002) ble_mesh_client: size_test_task: client publish i:9
W (285902) BLE_MESH: Ran out of retransmit attempts
I (313002) ble_mesh_client: size_test_task: client publish i:10
W (315902) BLE_MESH: Ran out of retransmit attempts
```

server model :
```
I (1886407) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (1886407) iot ble mesh server recv: 00 
I (2014367) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2014367) iot ble mesh server recv: 00 
I (2044367) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2044367) iot ble mesh server recv: 00 01 
I (2074387) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2074387) iot ble mesh server recv: 00 01 02 
I (2104367) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2104367) iot ble mesh server recv: 00 01 02 03 
I (2134387) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2134397) iot ble mesh server recv: 00 01 02 03 04 
I (2164367) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2164367) iot ble mesh server recv: 00 01 02 03 04 05 
I (2194367) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2194367) iot ble mesh server recv: 00 01 02 03 04 05 06 
I (2224367) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2224367) iot ble mesh server recv: 00 01 02 03 04 05 06 07 
W (2254457) BLE_MESH: Not sending ACK for non-unicast address
I (2254457) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2254467) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 
W (2254947) BLE_MESH: Got segment for already complete SDU
W (2254947) BLE_MESH: Not sending ACK for non-unicast address
W (2255027) BLE_MESH: Got segment for already complete SDU
W (2255027) BLE_MESH: Not sending ACK for non-unicast address
W (2255527) BLE_MESH: Got segment for already complete SDU
W (2255527) BLE_MESH: Not sending ACK for non-unicast address
W (2255607) BLE_MESH: Got segment for already complete SDU
W (2255607) BLE_MESH: Not sending ACK for non-unicast address
W (2256107) BLE_MESH: Got segment for already complete SDU
W (2256107) BLE_MESH: Not sending ACK for non-unicast address
W (2256247) BLE_MESH: Got segment for already complete SDU
W (2256247) BLE_MESH: Not sending ACK for non-unicast address
W (2256687) BLE_MESH: Got segment for already complete SDU
W (2256687) BLE_MESH: Not sending ACK for non-unicast address
W (2256767) BLE_MESH: Got segment for already complete SDU
W (2256767) BLE_MESH: Not sending ACK for non-unicast address
W (2284457) BLE_MESH: Not sending ACK for non-unicast address
I (2284457) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2284467) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 
W (2284947) BLE_MESH: Got segment for already complete SDU
W (2284947) BLE_MESH: Not sending ACK for non-unicast address
W (2285057) BLE_MESH: Got segment for already complete SDU
W (2285057) BLE_MESH: Not sending ACK for non-unicast address
W (2285537) BLE_MESH: Got segment for already complete SDU
W (2285537) BLE_MESH: Not sending ACK for non-unicast address
W (2285617) BLE_MESH: Got segment for already complete SDU
W (2285617) BLE_MESH: Not sending ACK for non-unicast address
W (2286107) BLE_MESH: Got segment for already complete SDU
W (2286107) BLE_MESH: Not sending ACK for non-unicast address
W (2286267) BLE_MESH: Got segment for already complete SDU
W (2286267) BLE_MESH: Not sending ACK for non-unicast address
W (2286707) BLE_MESH: Got segment for already complete SDU
W (2286707) BLE_MESH: Not sending ACK for non-unicast address
W (2286767) BLE_MESH: Got segment for already complete SDU
W (2286767) BLE_MESH: Not sending ACK for non-unicast address
W (2314477) BLE_MESH: Not sending ACK for non-unicast address
I (2314477) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (2314487) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 
W (2314957) BLE_MESH: Got segment for already complete SDU
W (2314957) BLE_MESH: Not sending ACK for non-unicast address
W (2315027) BLE_MESH: Got segment for already complete SDU
W (2315027) BLE_MESH: Not sending ACK for non-unicast address
W (2315527) BLE_MESH: Got segment for already complete SDU
W (2315527) BLE_MESH: Not sending ACK for non-unicast address
W (2315627) BLE_MESH: Got segment for already complete SDU
W (2315637) BLE_MESH: Not sending ACK for non-unicast address
W (2316107) BLE_MESH: Got segment for already complete SDU
W (2316107) BLE_MESH: Not sending ACK for non-unicast address
W (2316687) BLE_MESH: Got segment for already complete SDU
W (2316687) BLE_MESH: Not sending ACK for non-unicast address
W (2316767) BLE_MESH: Got segment for already complete SDU
W (2316767) BLE_MESH: Not sending ACK for non-unicast address
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3295735/sdkconfig.zip)

WCCWCC, 2019-06-17:
Hi zhanlej,

    This will not cause any problems.

    In the specification there is the following description：
> Note: A lower transport layer that receives a Segmented message that is addressed to a group orvirtual address does not start the acknowledgment timer and does not send Segment Acknowledgment messages.
    
     If you use unicast address , this will not happen.

zhanlej, 2019-06-18:
Hi @WCCWCC ,

I think the timeout period of publish retransmission is too long (about 3 seconds). And the number of retransmissions is too much. At the receiving end, it should be retransmitted 9 times.

When I narrowed the interval of the publish message from 30 seconds to 1 second, the receiver repeatedly received the previous data, which I think is incorrect.

Is there any way to shorten the retransmission interval and number of retransmissions of Publish?

```
I (133052) ble_mesh_client: size_test_task: client publish i:6
I (134062) ble_mesh_client: size_test_task: client publish i:7
I (135062) ble_mesh_client: size_test_task: client publish i:8
I (136062) ble_mesh_client: size_test_task: client publish i:9
I (137062) ble_mesh_client: size_test_task: client publish i:10
I (138062) ble_mesh_client: size_test_task: client publish i:11
W (139002) BLE_MESH: Ran out of retransmit attempts
I (139062) ble_mesh_client: size_test_task: client publish i:12
W (140002) BLE_MESH: Ran out of retransmit attempts
I (140062) ble_mesh_client: size_test_task: client publish i:13
W (141012) BLE_MESH: Ran out of retransmit attempts
I (141062) ble_mesh_client: size_test_task: client publish i:14
I (142062) ble_mesh_client: size_test_task: client publish i:15
W (142092) BLE_MESH: Ran out of retransmit attempts
I (143062) ble_mesh_client: size_test_task: client publish i:16
W (143312) BLE_MESH: Ran out of retransmit attempts
I (144062) ble_mesh_client: size_test_task: client publish i:17
I (145062) ble_mesh_client: size_test_task: client publish i:18
W (145202) BLE_MESH: Ran out of retransmit attempts
I (146062) ble_mesh_client: size_test_task: client publish i:19
W (146552) BLE_MESH: Ran out of retransmit attempts
I (147062) ble_mesh_client: size_test_task: client publish i:20
W (147932) BLE_MESH: Ran out of retransmit attempts
I (148062) ble_mesh_client: size_test_task: client publish i:21
I (149062) ble_mesh_client: size_test_task: client publish i:22
W (149822) BLE_MESH: Ran out of retransmit attempts
I (150062) ble_mesh_client: size_test_task: client publish i:23
I (151062) ble_mesh_client: size_test_task: client publish i:24
W (151982) BLE_MESH: Ran out of retransmit attempts
I (152062) ble_mesh_client: size_test_task: client publish i:25
I (153062) ble_mesh_client: size_test_task: client publish i:26
W (153602) BLE_MESH: Ran out of retransmit attempts
I (154062) ble_mesh_client: size_test_task: client publish i:27
I (155062) ble_mesh_client: size_test_task: client publish i:28
W (155312) BLE_MESH: Ran out of retransmit attempts
I (156062) ble_mesh_client: size_test_task: client publish i:29
W (156752) BLE_MESH: Ran out of retransmit attempts
I (157062) ble_mesh_client: size_test_task: client publish i:30
I (158062) ble_mesh_client: size_test_task: client publish i:31
I (159062) ble_mesh_client: size_test_task: client publish i:32
E (159062) BLE_MESH: send_seg, No multi-segment message contexts available
W (159312) BLE_MESH: Ran out of retransmit attempts
I (160062) ble_mesh_client: size_test_task: client publish i:33
W (160932) BLE_MESH: Ran out of retransmit attempts
I (161062) ble_mesh_client: size_test_task: client publish i:34
I (162062) ble_mesh_client: size_test_task: client publish i:35
E (162062) BLE_MESH: send_seg, No multi-segment message contexts available
W (162552) BLE_MESH: Ran out of retransmit attempts
I (163062) ble_mesh_client: size_test_task: client publish i:36
I (164062) ble_mesh_client: size_test_task: client publish i:37
E (164062) BLE_MESH: send_seg, No multi-segment message contexts available
```

```
I (228377) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (228377) iot ble mesh server recv: 00 01 02 03 04 05 
I (229387) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (229387) iot ble mesh server recv: 00 01 02 03 04 05 06 
I (230377) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (230387) iot ble mesh server recv: 00 01 02 03 04 05 06 07 
W (231867) BLE_MESH: Not sending ACK for non-unicast address
[... 311 lines cut ...]
I (256187) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (256187) iot ble mesh server recv: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 
W (256397) BLE_MESH: Not sending ACK for non-unicast address
I (256397) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (256397) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (256407) iot ble mesh server recv: 10 11 12 13 14 15 16 17 18 
W (256787) BLE_MESH: Not sending ACK for non-unicast address
I (256787) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (256787) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (256797) iot ble mesh server recv: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 
W (257117) BLE_MESH: Not sending ACK for non-unicast address
I (257117) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (257117) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (257127) iot ble mesh server recv: 10 11 12 13 14 15 16 17 18 19 1a 
W (257387) BLE_MESH: Not sending ACK for non-unicast address
I (257387) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (257387) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (257397) iot ble mesh server recv: 10 11 12 13 14 15 16 17 
W (257827) BLE_MESH: Duplicate SDU from src 0x0006
W (258127) BLE_MESH: Not sending ACK for non-unicast address
I (258127) ble_mesh_node: esp_ble_mesh_model_cb: IOT BLE MESH server receives msg, opcode 0xc002e5
I (258127) iot ble mesh server recv: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (258137) iot ble mesh server recv: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
I (258147) iot ble mesh server recv: 20
```

WCCWCC, 2019-06-18:
Hi zhanlej,

    We made a patch for this problem.You should enter your IDF path. run git am xxx.

     
[0001-component-ble_mesh-may-need-to-add-a-callback-for-fi.zip](https://github.com/espressif/esp-idf/files/3301429/0001-component-ble_mesh-may-need-to-add-a-callback-for-fi.zip)

zhanlej, 2019-06-19:
Hi @WCCWCC ,

Sorry, the patch you provided cannot be used, because I use the branch of v0.6.1. Some macro definitions are different from those in your patch, such as "BT_XXX" changed to "BLE_XXX". So I modified it manually.

WCCWCC, 2019-06-20:
Hi zhanlej，
  
    Has your previous problem been solved?

zhanlej, 2019-06-20:
Hi @WCCWCC ,
Thank you very much, the problem of repeatedly receiving data has been solved.

damgad, 2020-02-20:
Hey @Alvin1Zhang I have the same issue. Is the issue fixed in the current version? Seems like the patch from @WCCWCC is not incorporated into the sourcecode (none of the master, 4.0 and 4.1 branches have it). I guess that the issue should remain open until fixed.

Alvin1Zhang, 2020-02-21:
@damgad  Thanks for the notes. We will look into.

Campou, 2020-05-29:
Hi @damgad 

Sorry for the late reply.

The log level of `Not sending ACK for non-unicast address` has been changed to `INFO`. So when sending a segmented message to a non-unicast address, on the receiver side, no warning log like `Not sending ACK for non-unicast address` will be given.

Also did you meet the same problem of repeatedly receiving data from a segmented message? If yes, could you please help to provide the test log and test method?

Thanks.

Alvin1Zhang, 2020-07-23:
@damgad Any details could be shared per the suggestion by @Campou ? Thanks.

Alvin1Zhang, 2020-08-05:
@damgad hi, any details could be shared per the suggestion by @Campou ? Thanks.

Alvin1Zhang, 2020-09-09:
Thanks for the updates, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2020-09-09 by Alvin1Zhang

shiweichn, 2020-12-21:
I have the same problem.Node 1 sends data to Node 2, but node 2 receives duplicate data.
The IDF version I used was Release V4.1. How to solve this problem?

InfiniteYuan, 2020-12-22:
Thanks your feedback. Please take a try with this [patch](https://github.com/espressif/esp-idf/files/3301429/0001-component-ble_mesh-may-need-to-add-a-callback-for-fi.zip).

shiweichn, 2020-12-22:
> Thanks your feedback. Please take a try with this [patch](https://github.com/espressif/esp-idf/files/3301429/0001-component-ble_mesh-may-need-to-add-a-callback-for-fi.zip).

多谢回复，昨天我已经使用了这个patch进行测试，似乎已经解决了重复收到数据的问题。请问这个问题不算是一个bug吗？为什么不在release v4.x 版本中完善呢？