espressif/esp-idf #3643: examples/peripherals/rmt_nec_tx_rx don't work correct.

nopnop2002, 2019-06-17:
examples/peripherals/rmt_nec_tx_rx don't work correct.

## Environment

	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-512-g6fd535c98

## Problem Description
I added LOG to the cord.

	/*
	 * @brief Check whether this value represents an NEC header
	 */
	static bool nec_header_if(rmt_item32_t* item)
	{
	    ESP_LOGI(NEC_TAG, "item->level0=%d", item->level0);
	    ESP_LOGI(NEC_TAG, "item->level1=%d", item->level1);
	    ESP_LOGI(NEC_TAG, "item->duration0=%d", item->duration0);
	    ESP_LOGI(NEC_TAG, "item->duration1=%d", item->duration1);
	    if((item->level0 == RMT_RX_ACTIVE_LEVEL && item->level1 != RMT_RX_ACTIVE_LEVEL)
	        && nec_check_in_range(item->duration0, NEC_HEADER_HIGH_US, NEC_BIT_MARGIN)
	        && nec_check_in_range(item->duration1, NEC_HEADER_LOW_US, NEC_BIT_MARGIN)) {
	        return true;
	    }
	    return false;
	}

I launched an IR cord.
It's indicated as follows in esp - idf.

	I (181000) NEC: item->level0=0
	I (181000) NEC: item->level1=1
	I (181000) NEC: item->duration0=7186
	I (181000) NEC: item->duration1=3596
	I (181010) NEC: nec_bit(0) fail
	I (181010) NEC: nec_parse_items res=-1
	I (181050) NEC: xRingbufferReceive
	I (181050) NEC: nec_parse_items res=-1
	I (181880) NEC: xRingbufferReceive
	I (181880) NEC: item->level0=0
	I (181880) NEC: item->level1=1
	I (181880) NEC: item->duration0=7207
	I (181890) NEC: item->duration1=3581
	I (181890) NEC: nec_header_if fail
	I (181900) NEC: nec_parse_items res=-1
	I (181930) NEC: xRingbufferReceive
	I (181930) NEC: nec_parse_items res=-1

I tried Arduino Library.
It's indicated as follows in Arduino.

	Decoded NEC(1): Value:FFA857 Adrs:0
	Decoded NEC(1): Value:FFA857 Adrs:0 (32 bits) 
	Raw samples(68): Gap:27154
	  Head: m8850  s4500
	0:m500 s650	1:m450 s650		 2:m450 s650	3:m450 s700		 
	4:m450 s650	5:m450 s650		 6:m450 s650	7:m450 s650		 
	8:m500 s1750	9:m450 s1750		 10:m500 s1750	11:m450 s1750		 
	12:m450 s1750	13:m500 s1800		 14:m450 s1700	15:m500 s1750		 

	16:m450 s1800	17:m450 s600		 18:m500 s1800	19:m450 s650		 
	20:m450 s1750	21:m450 s650		 22:m450 s700	23:m450 s600		 
	24:m500 s600	25:m500 s1750		 26:m500 s600	27:m500 s1750		 
	28:m450 s650	29:m450 s1800		 30:m450 s1750	31:m450 s1750		 

	32:m500
	Extent=67200
	Mark  min:450	 max:500
	Space min:600	 max:1800

In Arduino, MARK is 8850, and SPACE is 4500.
Spec of NEC-Format:
MARK is 9000, SPACE is 4500.

In esp32, MARK is 7207, and SPACE is 3581 
duration0 of esp32 and duration1 of esp32 is too small.
Do you have any help.

r1dd1ck, 2019-06-17:
@nopnop2002 
> In esp32, Header MARK is 7207, and SPACE is 3581
> duration0 of esp32 and duration1 of esp32 is too small.

Those values are not in **μs** but in **RMT cycles** :exclamation:

and as the divider in the mentioned example is set to **100**, with a base RMT clock of **80MHz**, those values give:

(7207 * 100) / 80000000 = **~9009μs**
(3581 * 100) / 80000000 = **~4476μs**

which, I believe, is **close enough** to the NEC spec :wink:

nopnop2002, 2019-06-17:
OK.
Those values are not in μs but in RMT cycles.
I added LOG to nec_check_in_range();

	I (840) NEC: xRingbufferReceive
	I (840) NEC: item->level0=0
	I (840) NEC: item->level1=1
	I (840) NEC: item->duration0=7179
	I (850) NEC: item->duration1=3598
	I (850) NEC: duration_ticks=7179[RMT]-->8973[us]
	I (860) NEC: target_us=9000[us] margin_us=20[us]
	I (860) NEC: range(high)=9020[us] range(low)=8980[us]
	I (870) NEC: nec_header_if fail
	I (870) NEC: nec_parse_items res=-1
	I (900) NEC: xRingbufferReceive
	I (900) NEC: nec_parse_items res=-1

I change NEC_BIT_MARGIN from 20 to 100.
I got this.

	I (19860) NEC: RMT RCV --- addr: 0xff00 cmd: 0xea15

How are you thinking about nec_build_items()?

nopnop2002, 2019-06-18:
I confirmed NEC IR Specification.
NEC IR is begin from LSB.
So your example code is right.
Thank you for your help.
Please close this issues.
[image]

r1dd1ck, 2019-06-18:
I would suggest to change the divider in the example from 100 to 96. Then the pulses can be generated exactly on spec, with 7500 & 3750 RMT cycles representing the required 9ms & 4.5ms durations.

nopnop2002, 2019-06-18:
May i change this code?

	//#define RMT_CLK_DIV      100    /*!< RMT counter clock divider */
	#define RMT_CLK_DIV      96     /*!< RMT counter clock divider */

Thank you.

koobest, 2019-07-17:
@nopnop2002 
Yes, You can create a PR.

thanks !!

nopnop2002, 2019-07-17:
I close this issues.
Thanks for your support.

closed 2019-07-17 by nopnop2002