espressif/esp-idf #3644: Problem with esp_timer and ISR

filzek, 2019-06-17:
Latest MASTER as of  17 june 2019 
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0

The problem is that ISR and Timer had sometimes a conflict. 

When calling a esp_timer_start_once(xxxx, x) sometimes it brings this error  and crash:  

ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40085fa0

ELF file SHA256: b57df5f1784e06f16e1df78e780da87118183500edeaa80c3f97dce214cf904b

Backtrace: 0x40085b0c:0x3ffb0600 0x40085fa3:0x3ffb0620 0x4008466f:0x3ffb0640 0x400852dd:0x3ffb0670 0x4
0083fed:0x3ffb0690 0x400e62ef:0x3ffb5470 0x400d47cb:0x3ffb5490 0x40087be6:0x3ffb54b0 0x40088419:0x3ffb
54d0
0x40085b0c: invoke_abort at /esp/esp-idf2/components/esp32/panic.c:156

0x40085fa3: _esp_error_check_failed at /esp/esp-idf2/components/esp32/panic.c:721

0x400852dd: gpio_isr_loop at /esp/esp-idf2/components/driver/gpio.c:334
 (inlined by) gpio_intr_service at /esp/esp-idf2/components/driver/gpio.c:349

0x40083fed: _xt_lowint1 at /esp/esp-idf2/components/freertos/xtensa_vectors.S:1153

0x400e62ef: esp_pm_impl_waiti at /esp/esp-idf2/components/esp32/pm_esp32.c:493

0x400d47cb: esp_vApplicationIdleHook at /esp/esp-idf2/components/esp_common/src/fr
eertos_hooks.c:63

0x40087be6: prvIdleTask at /esp/esp-idf2/components/freertos/tasks.c:3383 (discrim
inator 1)

0x40088419: vPortTaskWrapper at /esp/esp-idf2/components/freertos/port.c:143

Rebooting...

*********************
The GPIO imput runs at 120 ISR per second, time framed perfect in  esp_timer_get_time() and in scope, there is no faulting frames, so, the esp_timer_start_once() will be called 120 times per second.

AFIK there is some problem with ISR calling esp_timer_start_once so frenquently and those bring the error. 

I think this is a case for Igor to check as maybe an issue in GCC or internal ESP routines.

igrr, 2019-06-17:
Can you describe the relationship between the timer and the GPIO interrupt? The error code indicates that you are trying to start a timer which is still running (i.e. not elapsed yet).

If you want to restart the timer from a GPIO interrupt, you may need to call esp_timer_stop first, to ensure that the timer is not running.

filzek, 2019-06-18:
Hi Ivan @igrr  we have found out that sometimes even after the timer has been dispatch it still armed in memory - like it was never trigger, so we add a failsafe switch using a variable to check the timer dispatch and if not we just stop timer and rearm it. Seens to solve the problem but not the cause it happen.

Also, now we face a new problem with the same behavior.

Esp timer and wifi.

When we enable WiFi in ESP32 the esp timer become non functional, something change states and gpios behavior non-linear.

Functions runs perfect with ISR enable timers, dispatch perfect.

Enable WiFi.

Timers do not behavior as before and everything is not working.

filzek, 2019-06-18:
> Can you describe the relationship between the timer and the GPIO interrupt? The error code indicates that you are trying to start a timer which is still running (i.e. not elapsed yet).
>
> If you want [...]

It seens that all problem stay when WiFi is enable

When ISR register an interrupt with WiFi Off:

TIME           | OFFSET
419345360 |  
419353710 | 8350
419362036 | 8326
419370388 | 8352
419378710 | 8322
419387047 | 8337
419395360 | 8313
419403706 | 8346
419412045 | 8339
419420395 | 8350
419428700 | 8305
419437038 | 8338
419445371 | 8333
419453736 | 8365
419462049 | 8313
419470369 | 8320
419478704 | 8335

WiFi On
TIME           | OFFSET
420287109 | 41706
420387094 | 99985
420487107 | 100013
420587104 | 99997
420687125 | 100021
420787135 | 100010
420887129 | 99994
420987128 | 99999
421087149 | 100021
421187156 | 100007
421287173 | 100017
421387144 | 99971
421487171 | 100027
421587193 | 100022
421687185 | 99992
421787178 | 99993
421887198 | 100020
421987203 | 100005
422087227 | 100024
422187225 | 99998
422287224 | 99999
422387242 | 100018
422487247 | 100005
422587275 | 100028
422687261 | 99986
422787265 | 100004
422887294 | 100029
422987293 | 99999
423087297 | 100004
423187312 | 100015
423287313 | 100001
423387337 | 100024
423487326 | 99989
423587341 | 100015
423687191 | 99850
423787365 | 100174
423887389 | 100024
423987375 | 99986
424087376 | 100001
424187413 | 100037
424287401 | 99988

So, in the scope, all is very clear and perfect, but in the ESP32 the ISR cant register correct anymore in US level. and ISR resolution seens to drop from 1us (sure to be in between 30us and 1000us minimun) to rougly 100000us.

Why the ISR behavior like this when WiFi is on?

igrr, 2019-06-18:
Could you please explain the relation between esp_timer and "ISR" in your program? Are you referring to the timer ISR or some other ISR?
Can you roughly describe what the program is doing? This will help reproduce the issue. Thanks.

filzek, 2019-06-25:
> Could you please explain the relation between esp_timer and "ISR" in your program? Are you referring to the timer ISR or some other ISR?
> Can you roughly describe what the program is doing? This will help reproduce the issue. Thanks.

Hi @igrr 

We found out how it happen.

We set 2 hardware timers HW0_0 and HW0_1 and an GPIO ISR.
Then GPIO ISR alarm, it run HW0_0 for 300 us, and   HW0_1 for 380us (HW0_0+80us).
HW0_0 will alarm a GPIO Level High.
HW0_0 will alarm a GPIO Level Low.

So, in Scope we can see all signals.

Problem is: 
ISR GPIO sometime will not dispatch, so it just do not ack the true ISR. 

HW timers are very precise, but sometimes it just also not dispare or dispare in an wrong time frame, just if WIFI is enable, if WiFi is disable this doenst happen.

3rd problem)
We cant use a ESP_TIMER for any other function because it will dispatch in an wierdo way folling the GPIO ISR.

igrr, 2019-08-18:
@filzek Are you trying to implement TRIAC control or something similar? If you can share the code which reproduces the issue ("ISR GPIO sometime will not dispatch"), we can investigate. In general, if you see that this happens when Wi-Fi is active, i can recommend 2 things:
1. Register the GPIO ISR on APP CPU (create a task pinned to APP CPU, call gpio_isr_register or gpio_install_isr_service there).
2. Raise the level of GPIO ISR to level 3. This can be done by passing `ESP_INTR_FLAG_LEVEL3 | ESP_INTR_FLAG_IRAM ` as the intr_alloc_flags argument to the above mentioned functions.

With regards to the other problem ("cant use a ESP_TIMER for any other function because it will dispatch in an wierdo way folling the GPIO ISR"), sorry, without a more technical description of the issue and/or some sample which shows what you mean by "weirdo", i don't think we can help.

filzek, 2021-01-05:
In Esp idf v4.2 it was fixed, so the code run perfect, something inside sdk was fixed.

closed 2021-01-05 by filzek