espressif/esp-idf #3647: Base64 decoded string not zero terminated?

steelcuts, 2019-06-18:
Hello,

had some weird returned strings lately while decoding base64 encoded strings.

[Related file](https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/src/wpa2/utils/base64.c)

I think there is missing something like `*pos = '\0';` at line 152, however the returned size_t pointer value seems to be ok so far.

Greetings :)

sagb2015, 2019-06-18:
@steelcuts - The function definition is inline with its API description and also resembles upstream supplicant code [here](https://w1.fi/cgit/hostap/tree/src/utils/base64.c?id=0f3d578efcff5f29d3b8af13edc3c1a9a3222129).  The return value should be treated as byte buffer of size _out_len_ rather than null terminated string.

sagb2015, 2019-07-11:
@steelcuts - Closing this issue. Feel free to re-open in case of doubts or unexpected behavior.

sagb2015, 2019-10-11:
This does not seem to require any fixing. Feel free to open again if required.

closed 2019-10-11 by sagb2015