espressif/esp-idf #3651: Generating binaries in Windows and Linux environment

hvishu, 2019-06-18:
Could the binaries generated in windows environment and Linux environment are comparable? I generated the application with the same application (same IDF version), I see only partitions.bin are matching, not the bootloader.bin or the app.bin.

We are debugging this issue https://github.com/espressif/esp-idf/issues/3500, which is seen in binaries generated out of windows environment and not yet seen in those generated in Linux environment so far.   Hence the question.

Regards,
Vishwa

hvishu, 2019-07-07:
Any updates for me on this?

projectgus, 2019-08-01:
Hi @hvishu ,

Sorry for the slow reply.

There are differences between binaries generated on different computers. If nothing else, some parts of ESP-IDF use the `__FILE__` macro which means the binary may contain the full path to a file. Also, some compilation and linking orders depend on the order that the filesystem returns paths to the build system.

The new CMake build system is a little more consistent in this regard, but there will still be differences in the binaries built on different computers.

The main practical difference between these is flash cache behaviour which may influence timing - because the ESP32 caches flash cache contents of .rodata and instructions, moving the addresses of these around in different binaries may cause more/less cache misses at particular times and subtly change the timing of some functions.

So if you have a bug which only appears on binaries generated from some machines, the root cause is most likely that the firmware has a race condition but the timing that triggers this race condition is only exposed with some patterns of flash access.

closed 2019-08-01 by projectgus