espressif/esp-idf #3653: Custom Partition Will NOT Build and always say overlap

filzek, 2019-06-18:
Latest SDK as of:
GCC 8.2 2019r1
IDF v4.0-dev-837-g58df1d93b

Cant build the system anymore, custom partition will always say it overlap, no matter what you try to do, just cant fix from here.

Error:
Building partitions partitions.csv...
CSV Error: First partition offset 0x9000 overlaps end of partition table 0xa000

You can just grab any partition example and try to build very same behavior.

Custom partition using TWO OTA slots as for reference:

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,            data,   nvs,        0x9000,     0x4000
otadata,        data,   ota,        0xd000,     0x2000
phy_init,       data,   phy,        0xf000,     0x1000
ota_0,          0,      ota_0,      0x10000,    0x180000
ota_1,          0,      ota_1,      0x190000,   0x180000
factory_nvs,    data,   nvs,        0x310000,   0x4000
nvs_keys,       data,   nvs_keys,   0x314000,   0x1000
storage,        data,   fat,        0x315000,   0xEB000

igrr, 2019-06-18:
What is the value of CONFIG_PARTITION_TABLE_OFFSET in your sdkconfig? (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-partition-table-offset)

KonstantinKondrashov, 2019-06-19:
> To have each partition offset adapt to the configured partition table offset, leave all partition offsets blank in the CSV file.

```
nvs, data, nvs, , 0x4000
otadata, data, ota, , 0x2000
phy_init, data, phy, , 0x1000
ota_0, 0, ota_0, , 0x180000
ota_1, 0, ota_1, , 0x180000
factory_nvs, data, nvs, , 0x4000
nvs_keys, data, nvs_keys, , 0x1000
storage, data, fat, , 0xEB000
```

Alvin1Zhang, 2019-07-10:
@filzek Thanks for reporting the issue, would you please help share if your issue has been resolved? Thanks.

closed 2019-08-04 by filzek

vinothkrstatiq, 2022-06-12:
Hi i also faced similar issue 
but i changed the value of CONFIG_PARTITION_TABLE_OFFSET to Default value:
“0x8000” it solved my problem 
in csv file
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, spiffs,  ,        0xF0000,
below i used 
@igrr Thanks for the help