espressif/esp-idf #3654: Interrupt UART_GLITCH_DET_INT is not triggered on the START bit in RS485 mode

neorevx, 2019-06-18:
## Environment

- Module or chip used:  ESP32-WROOM-32
- Operating System: Windows
- Power Supply: USB

## Problem Description

Interrupt UART_GLITCH_DET_INT is not triggered on the START bit in RS485 mode.

### Expected Behavior

When start receiving a byte, interrupt UART_GLICTH should be triggered.
From reference manual:

> UART_GLITCH_DET_INT: Triggered when the receiver detects a START bit

### Actual Behavior

Glitch interruption is apparently triggered when there is noise on the bus. Or maybe is triggered once.

### Steps to repropduce

Enable UART_GLITCH_DET_INT_ENA in interrupt mask and check interrupt.

koobest, 2019-06-19:
Hi, @neorevx 
This interrupt is generated when the receiver detects glitch at the start bit. This is a problem with the documentation. we will correct documentation.

thanks !!

neorevx, 2019-06-19:
Hi @koobest,

thank you. I'll check out your comment and close this issue.

closed 2019-06-19 by neorevx