espressif/esp-idf #3657: ESP32: Power On status of the pins during boot up process.

Yuvaraj4041, 2019-06-19:
Hi Guys,

My project uses GPIO pins to drive PWM to control power to a load. The issue I am facing now is that the power-on-default value of the pins is driving the MOSFET fully on - this will damage my load as the current is uncontrolled. This condition exists until the boot up process is completed and firmware change the status of the selected pin.

Since the boot time is too large, we need a way to instruct the ESP32 MCU to keep the pins in the required condition during power-on / boot up process.

Can anyone please let me know how to do this?

koobest, 2019-06-19:
Hi, @Yuvaraj4041 
Which GPIO are used?  you can refer to A.4. IO_MUX chapter of the ESP32 Datasheet to find the default mode of the GPIO when power on, and then, select the GPIO to control MOSFET. also you can use external weak pull-up or pull-down to set the IO to a default level.

thanks !!

Yuvaraj4041, 2019-06-19:
HI @koobest 

I am using GPIO 18 and GPIO19 as LED PWM.

koobest, 2019-06-19:
Hi, @Yuvaraj4041 
`OE-0 / IE - 0` means output /input is disabled,  `OE-1 /IE - 1` means output /input is enabled.
`wpu` means weak pull up, `wpd` means weak pull down,  if `wpu` and `wpd` is not mentioned, it means  the GPIO is floating when reset or after reset. 

What output level do you want when the chip is reset? you can use a  external  pull-up/pull-down resistors(about  40K). If the level is high, when configure the LEDC, you need to do the following processing:
```
ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TIMER_13_BIT,
        .freq_hz = 5000,
        .speed_mode = LEDC_HIGH_SPEED_MODE,
        .timer_num = LEDC_TIMER_0,
    };
    ledc_channel_config_t ledc_channel = {
        .channel    = LEDC_HS_CH0_CHANNEL,
        .duty       = 0x1 << (LEDC_TIMER_13_BIT),
        .gpio_num   = GPIO_NUM_18,
        .speed_mode = LEDC_HIGH_SPEED_MODE,
        .hpoint     = 0,
        .timer_sel  = LEDC_TIMER_0,
    };
    ledc_timer_config(&ledc_timer);
    ledc_stop(LEDC_HS_MODE, LEDC_HS_CH0_CHANNEL, 1);
    ledc_channel_config(&ledc_channel);
    ledc_update_duty(LEDC_HS_MODE, LEDC_HS_CH0_CHANNEL);
```

thanks !!

Yuvaraj4041, 2019-06-19:
Thanks @koobest 

I want **0** as output level when the chip is reset. so i think i have to use pull down resister for GPIO18 and GPIO19. 

If i select gpio with **wpd**, i think i dont need any pull down resistor and this will not drive my MOSFET when chip is boot up.

koobest, 2019-06-19:
Did GPIO18 / GPIO19 output  high level during reset?

Yuvaraj4041, 2019-06-19:
By default it is giving 0.07v, when i reset the chip then also same 0.07v

koobest, 2019-06-19:
Did this cause  the MOSFET fully on issue?

rkashi, 2019-06-20:
Hi koobest,

The load is an LED Light, its driven by a discrete buck converter with an N-Channel MOSFET and inverting gate driver using transistors. When the GPIO is low, the MOSFET gate is driven - causing an unregulated current to flow. 

>Reset condition OE-0, IE-0, after Reset condition OE-0, IE-1. 
This means that, it is tri-state at reset and after reset, this pin is configured as input.  So, most likely a weak pull up would solve the issue.

Question: Is there a start up file, where we can insert assembly code to configure the pins as output and drive them high?

koobest, 2019-06-20:
You can do this in `bootloader_init`, but this is not a good solution, I think.

rkashi, 2019-06-20:
Hello Koobest, 

Thank you I will try this out and let you know.
Question: what is the meaning of this new field  **.hpoint     = 0,**

``ledc_channel_config_t ledc_channel = {
        .channel    = LEDC_HS_CH0_CHANNEL,
        .duty       = 0x1 << (LEDC_TIMER_13_BIT),
        .gpio_num   = GPIO_NUM_18,
        .speed_mode = LEDC_HIGH_SPEED_MODE,
        **.hpoint     = 0,**
        .timer_sel  = LEDC_TIMER_0,

koobest, 2019-06-20:
Hi,
Please see the diagram. When the timer counter reach `hpoint`  value, the LEDC output high. 

[image]

Alvin1Zhang, 2019-07-02:
@Yuvaraj4041  Thanks for reporting the issue, feel free to reopen if your issue still exists. Thanks.

closed 2019-07-02 by Alvin1Zhang

raako71, 2021-09-17:
> e ESP32 Datasheet to find the default mode of the GPIO when power on, and then, select the GPIO to control MOSFET. also you can use external weak pull-up or pull-down to set the IO to a default level.

how can one interpret this part of the data sheet, which part explains boot status of the pin?