espressif/esp-idf #3658: What function is used to restart esp32

Herwey, 2019-06-19:
Hi :
Now , I want to restart the esp32 with fucntions, not by click the reset botton. So which function can help me ?
Thanks in advance !

Herwey, 2019-06-20:
I used these codes:
                fflush(stdout);//使stdout清空，就会立刻输出所有在缓冲区的内容。
                esp_restart();//芯片复位 函数位于esp_system.h
They can reset esp32, do you know them function, which reset is ?

Dazza0, 2019-06-20:
`esp_restart()` does a CPU reset, but also resets some special peripherals (e.g. WiFi and BT) ([see API reference here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/system.html?highlight=esp_restart#_CPPv411esp_restartv)). 

> They can reset esp32, do you know them function, which reset is ?

Again, it depends on what type of reset you are aiming to achieve. `esp_restart()` will do a CPU reset. Is this your intended goal?

Alvin1Zhang, 2019-06-21:
@Herwey Thanks for reporting the issue. If you have further such questions, please help post on our [forum](https://esp32.com), thanks.

closed 2019-06-21 by Alvin1Zhang