espressif/esp-idf #3660: mDNS does not send goodbye messages when removing services

KidVizious, 2019-06-19:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.0-dev-512-g6fd535c98
- Build System:         CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2018r1) 8.2.0
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description

When calling mdns_free() or mdns_remove_all_services() an mDNS goodbye message is not sent.  Using an iOS device, I can see it generate announcements when it wakes up and goodbyes when it goes to sleep in WireShark.  Upon mDNS init, the ESP32 sends the announcesments for all configured services as can be observed in WireShark.  mDNS discovery tools such as avahi-discover or dns-sd also show the presence of the ESP.  

However, when mDNS services are removed or the mDNS service is stopped using mdns_free(), goodbye messages are not seen on the network and ESP devices still show up in avahi-discover and dns-sd

### Expected Behavior
Goodbye messages are sent when shutting down mDNS and devices disappear from avahi-discover.

### Actual Behavior
No goodbye messages are seen, and avahi-discover still shows devices in list.

### Steps to repropduce

1. Init mDNS stack and add services (services will be discovered on network and mDNS advertisedments seen in WireShark
2. call mdns_free (services still advertiesed, no goodbye messages seen)

lucascoxBAF, 2019-06-20:
@liuzfesp @Alvin1Zhang , Any idea if there are plans to fix this soon? I have also noticed that there is no way to send "Announcements to Flush Outdated Cache Entries" per [rfc 67662 section 10.2](https://tools.ietf.org/html/rfc6762#section-10.2)

me-no-dev, 2019-06-20:
I would guess that there is not enough time for the messages to be sent. Code is there though. If you remove the services before freeing the service, bye should be sent. It is implemented.

me-no-dev, 2019-06-20:
Here is the method that actually does it. You can trace it in the code and see how it's organised. https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L1432

david-cermak, 2019-06-20:
Let me just note a workaround to have the bye packet sent on `mdns_remove_all_services()`. It is possible to create a service with null instance, in that case a default instance name is used as service instance name:
```
    //set default mDNS instance name
    ESP_ERROR_CHECK( mdns_instance_name_set("ESP32-WebServer") );
    //initialize service
    ESP_ERROR_CHECK( mdns_service_add(NULL, "_http", "_tcp", 80, serviceTxtData, 3) );
```

The issue seems to be that the library tends to skip the function @me-no-dev referenced if the service has instance name configured. This should be fixed shortly.

pete-maley, 2019-07-12:
Has there been any progress on this issue?

david-cermak, 2019-07-14:
The fix is currently under review in the internal repository, sorry for the delay

pete-maley, 2019-08-08:
Hello, just following up on this issue.  Last update was that it was under internal review 25 days ago.  Appreciate any updates you can provide.

closed 2019-08-21 by igrr