espressif/esp-idf #3662: Extra PSRAM on the heap is available for malloc() but not for new()

Jeroen88, 2019-06-19:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v3.2-111-g524308c98
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I can allocate a large block (900,000 uint8_t) with malloc() but not with new()

### Expected Behavior

Because of the available PSRAM to the heap, I would expect new() to be able to allocate the block of memory

### Actual Behavior

new() crashes the ESP

### Steps to repropduce

1. just flash and run the attached code

### Code to reproduce this issue

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\nStarted\n");
  Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  Serial.printf("SPIRam Total heap %d, SPIRam Free Heap %d\n", ESP.getPsramSize(), ESP.getFreePsram()); 
  Serial.printf("ChipRevision %d, Cpu Freq %d, SDK Version %s\n",ESP.getChipRevision(), ESP.getCpuFreqMHz(), ESP.getSdkVersion());
  Serial.printf("Flash Size %d, Flash Speed %d\n",ESP.getFlashChipSize(), ESP.getFlashChipSpeed());
#if defined(BOARD_HAS_PSRAM)
  Serial.printf("Board has PSRAM\n");
#else
  Serial.printf("Board doesn't have PSRAM\n");
#endif

  Serial.printf("Test malloc()...\n");
  uint8_t *testMallocBuffer = (uint8_t*)ps_malloc(900000);
  if(testMallocBuffer != NULL) Serial.printf("NOT NULL\n"); else Serial.printf("NULL\n");
  free(testMallocBuffer);

  Serial.printf("Test new()...\n");
  uint8_t *testNewBuffer = new uint8_t[900000];
  if(testNewBuffer) Serial.printf("test NOT NULL\n"); else Serial.printf("test NULL\n");
  delete testNewBuffer;
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Started

Internal Total heap 378424, internal Free Heap 348176
SPIRam Total heap 4194252, SPIRam Free Heap 4194252
ChipRevision 1, Cpu Freq 240, SDK Version v3.2-111-g524308c98
Flash Size 4194304, Flash Speed 80000000
Board has PSRAM
Test malloc()...
NOT NULL
Test new()...
abort() was called at PC 0x400e9163 on core 1

Backtrace: 0x4008dce8:0x3ffb1ea0 0x4008df15:0x3ffb1ec0 0x400e9163:0x3ffb1ee0 0x400e91aa:0x3ffb1f00 0x400e8bd7:0x3ffb1f20 0x400e8dfa:0x3ffb1f40 0x400e8db1:0x3ffb1f60 0x400d0dec:0x3ffb1f80 0x400d209f:0x3ffb1fb0 0x4008c681:0x3ffb1fd0

Rebooting...

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

## Decoding stack results

```

0x4008dce8: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008df15: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400e9163: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x400e91aa: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x400e8bd7: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x400e8dfa: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400e8db1: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400d0dec: setup() at /home/jeroen/Arduino/Test_PSRAM_malloc_new/Test_PSRAM_malloc_new.ino line 21
0x400d209f: loopTask(void*) at /home/jeroen/Downloads/arduino-1.8.9-linux64/arduino-1.8.9/hardware/espressif/esp32/cores/esp32/main.cpp line 14
0x4008c681: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

negativekelvin, 2019-06-19:
ps_malloc specifically targets psram
https://github.com/espressif/arduino-esp32/blob/010a7c60f7000eb17c6333309c87156769848a0e/cores/esp32/esp32-hal-psram.c#L55-L60
malloc has to be configured to include psram
https://github.com/espressif/esp-idf/blob/2c55fae6cf8de7f9776840b533acfc31e2de1b24/components/esp32/Kconfig#L63-L64

https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/tools/sdk/sdkconfig#L271

projectgus, 2019-06-20:
Hi @Jeroen88 ,

Yes, @negativekelvin's assessment is correct. This is a choice in the Arduino project's configuration of ESP-IDF, not a limitation of ESP-IDF.

Feel free to open an issue on the arduino-esp32 project to ask about changing it. However there are other implications for the change so the Arduino maintainers may not be keen to do so.

You can also build Arduino as a component in ESP-IDF, in which case you can change the configuration yourself.

Because this is C++ another option would be to override the `new` operator in the places you need, in order to call ps_malloc() internally instead of malloc().

Angus

closed 2019-06-20 by projectgus

Jeroen88, 2019-06-20:
@negativekelvin @projectgus thank you both, however, a compiler flag has been defined in the Arduino environment in boards.txt

``
esp32wrover.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue
``

And 'plain' malloc() indeed behaves like SPIRAM_USE_MALLOC. However, my problem is that new() does not behave the same: **new() does not make use of the PSRAM**

In my opinion this is an error, both malloc() and new() should follow the same behavior if SPIRAM_USE_MALLOC is set.

negativekelvin, 2019-06-20:
Why did you call ps_malloc instead of malloc in your example?

Jeroen88, 2019-06-20:
Thnx @negativekelvin, that was not my intention. So now I am not sure if a plain malloc() works. I will try that later today...

Jeroen88, 2019-06-20:
@negativekelvin using malloc() the pointer is NULL, so it won't crash (at that point) but PSRAM is not made available to the normal heap. new() still crashes. I am going to investigate further. Thnx for your help!

igrr, 2019-06-20:
It doesn't crash, it tries to throw a bad_alloc exception, and fails because Arduino-esp32 has been compiled with zero emergency exception pool size. The standard behavior is to call std::terminate in this case.

Jeroen88, 2019-06-20:
Thnx @igrr you are right!