espressif/esp-idf #3663: WiFi stack can freeze if CONFIG_ESP32_WIFI_RX_BA_WIN > CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM

ssilverman, 2019-06-20:
Suggest you change the issue to this:

> WiFi stack can freeze if CONFIG_ESP32_WIFI_RX_BA_WIN > CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM.
> 
> Tested on esp-idf v3.2
> 
> This configuration should be prevented or checked and an error thrown.

_Originally posted by @negativekelvin in https://github.com/espressif/esp-idf/issues/3646#issuecomment-503832957_

ssilverman, 2019-06-20:
See also: https://github.com/espressif/arduino-esp32/issues/2899

r1dd1ck, 2019-06-20:
Well, are you sure it's this?
 
Because the `iperf` example uses:
`WIFI_STATIC_RX_BUFFER_NUM=16 `
`WIFI_RX_BA_WIN=32`

and it does **not** "freeze" :neutral_face:

negativekelvin, 2019-06-20:
Well it solves the problem reported here but maybe only happens "under certain conditions" so needs more constraints from wifi stack developers

r1dd1ck, 2019-06-20:
Yes, I certainly agree that changing the values "fixes" the reported issue :thumbsup: 

But what I was pointing at is, whether the supplied conditional really is the **underlying cause** of the freezing stack, because it is in direct conflict with the `iperf` example defaults (which doesn't seem to show the symptoms). 

Other than that, all is fine :slightly_smiling_face:

ssilverman, 2019-09-26:
This still happens in the v1.0.3 Arduino core: https://github.com/espressif/arduino-esp32/issues/3287

ssilverman, 2019-09-26:
For those following along, the stack now fails to receive more packets than defined by CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM. Doing some spelunking for the word `dynamic_rx_buf_num` (the consumer of this define, as found with a simple grep) led me to `tools/sdk/lib/libesp32.a`, that's why I think the issue is in `esp-idf` as opposed to just in the Arduino core.

espxiehang, 2019-11-14:
@ssilverman hi,I use your script to send upd packets, but I didn't reproduce the issue.
Can you give me an easy to reproduce method and your log?

ssilverman, 2019-11-14:
When I filed this and related issues, this was the behaviour. However, the later Arduino releases seem to fix the problem. I haven’t dived in, but if you use settings from 1.0.2 (I think? See the issues in that repo that link to this one.), then you’ll see this behaviour. I realize this repo isn’t Arduino, and I haven’t explored much in the non-Arduino space of the esp-idf repo, but I think you’ll find the behaviour by using the settings mentioned in those Arduino versions.

negativekelvin, 2019-11-14:
I think this can be closed with https://github.com/espressif/esp-idf/commit/dd26caf7799a7414596d71d1bc4ee39634480a62#diff-fdc67a5904b65c3885aad20459aa1cef

ssilverman, 2019-11-14:
@negativekelvin was the source of this problem found?

negativekelvin, 2019-11-15:
Yes, enforce restrictions on the value of CONFIG_ESP32_WIFI_RX_BA_WIN per the commit.  The other issues are all outside the scope of esp-idf code so I think you can close this.

ssilverman, 2019-11-15:
Right, but I meant: why does not meeting those conditions cause the problem? Is there a deeper problem somewhere? Should a comment in the code that uses these values make note of a fault potential? I’m just suggesting another commit that addresses that problem or at least makes a note of it somewhere. Other than that, I can certainly close this if you’re saying it’s fixed.

closed 2019-11-15 by ssilverman

negativekelvin, 2019-11-15:
The actual conditions in the commit are 

`(CONFIG_ESP32_WIFI_RX_BA_WIN ≤ CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM) && (CONFIG_ESP32_WIFI_RX_BA_WIN ≤ 2 * CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM)`

so iperf is not in violation

espxiehang, 2019-11-15:
Yes, its rules are defined in “components/esp_wifi/src/wifi_init.c”