espressif/esp-idf #3666: SYSTEM_EVENT_STA_DISCONNECTED with error code 108

farizadp, 2019-06-20:
## Environment

- Module or chip used:  ESP32-PICO-D4
- IDF version ): 8e51f7e2363d613fdcea48707139cdc20b141832 release/v3.2
- Build System:         Make
- Compiler version:    crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         external 5V

## Problem Description

```esp_wifi_connect()``` with callback, results in an event ```SYSTEM_EVENT_STA_DISCONNECTED``` with reason 108. I cannot find error 108 in ```esp_wifi_types.h```.  Before the event is triggered, I see message from wifi lib ```wifi: ampdu: ignore deleting tx BA0```

Anyone can explain this behavior?

farizadp, 2019-06-20:
Ok, I managed to reduce the occurrences of ampdu messages. Below is what I get. So it's connected to but after a while it disconnects.

```
I (57391) wifi: n:1 1, o:1 0, ap:1 1, sta:1 0, prof:1
I (59087) wifi: state: init -> auth (b0)
I (59094) wifi: state: auth -> assoc (0)
I (59101) wifi: state: assoc -> run (10)
I (59116) wifi: connected with PD_Test, channel 1
I (59117) wifi: pm start, type: 1

I (59120) p1_app: wifi_event_handler()[726][111284]: SYSTEM_EVENT_STA_CONNECTED
I (59121) p1_app: wifi_event_handler()[733][111284]: Authentication mode (3)(4): wpa2Psk
I (84034) wifi: state: run -> init (6cc0)
I (84035) wifi: pm stop, total sleep time: 16678451 us / 24917551 us

I (84036) wifi: n:1 0, o:1 1, ap:1 1, sta:1 0, prof:1
I (84041) p1_app: wifi_event_handler()[736][112072]: SYSTEM_EVENT_STA_DISCONNECTED
I (84046) p1_app: wifi_event_handler()[741][112072]: Disconnected from: PD_Test
I (84054) p1_app: wifi_event_handler()[743][112072]: Reason: (108)
```

zhangyanjiaoesp, 2019-06-21:
@farizadp The dauth reason 108 may be defined by the router itself. Would you please provide the following info:
1. Provide the capture packages when this issue happen.
2. Which AP did you use?
3. How to reproduce your problem?

farizadp, 2019-06-25:
@zhangyanjiaoesp : It will take sometime to provide you with those information. I will get back to you with the information as soon as possible

Meanwhile, what does the following message means?
```
I (227678) wifi: ampdu: ignore deleting tx BA0
```

Below is my sdkconfig.h
```
$ cat build/include/sdkconfig.h | grep '_TX\|AMPDU'
#define CONFIG_ESP32_PHY_MAX_TX_POWER 20
#define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER 1
#define CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER 20
#define CONFIG_ESP32_WIFI_TX_BA_WIN 6
#define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM 16
#define CONFIG_DMA_TX_BUF_NUM 10
#define CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED 1
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 1
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
```

zhangyanjiaoesp, 2019-07-01:
@farizadp 
`I (227678) wifi: ampdu: ignore deleting tx BA0` means ESP32 receives a `Delete BA request` from the route, ESP32 will not delete the BA at present, but we will fix this issue in the later version. 
For your current problem, we need your capture packages for further analysis.

Alvin1Zhang, 2019-07-17:
@farizadp Thanks for reporting the issue, do you get a chance to capture the packets? Thanks.

chegewara, 2019-07-29:
Hi,
im having the same issue with arduino as component. Basically i want to test what is causing when i am downloading files and for tests i am downloading ~40kB file from LAN NAS in loop. After some downloads im getting this  `wifi: ampdu: ignore deleting tx BA0` and this is log before app stop responding:
```
I (265757) wifi: ampdu: ignore deleting tx BA0
I (271757) wifi: ampdu: ignore deleting tx BA0
I (288787) wifi: ampdu: ignore deleting tx BA0
I (351877) wifi: ampdu: ignore deleting tx BA0
I (449017) wifi: ampdu: ignore deleting tx BA0
I (700497) wifi: state: run -> init (7c0)
I (700497) wifi: pm stop, total sleep time: 0 us / 692360439 us

I (700497) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (820497) event: station ip lost
```
Disabling AMPDU in menuconfig does not help too much, no ampdu message in logs, but wifi stop responding anyway.

chegewara, 2019-07-30:
I also have wireshark file from esp32 and similar request from PC browser. 
```
0000   26 18 1d 97 ea 0a 96 e9 62 ef 34 f3 08 00 45 00   &...ê..ébï4ó..E.
0010   00 85 00 0a 00 00 fd 06 80 32 c0 a8 2b 29 32 74   ......ý..2À¨+)2t
0020   1e f1 f0 8d 00 50 00 00 19 6e 9d d1 c1 24 50 18   .ñð..P...n.ÑÁ$P.
0030   43 50 5a 02 00 00 47 45 54 20 2f 4d 41 49 58 2f   CPZ...GET /MAIX/
0040   4d 61 69 78 50 79 2f 61 73 73 65 74 73 2f 41 6c   MaixPy/assets/Al
0050   69 63 65 2e 62 6d 70 20 48 54 54 50 2f 31 2e 31   ice.bmp HTTP/1.1
0060   0d 0a 48 6f 73 74 3a 20 64 6c 2e 73 69 70 65 65   ..Host: dl.sipee
0070   64 2e 63 6f 6d 0d 0a 63 61 63 68 65 2d 63 6f 6e   d.com..cache-con
0080   74 72 6f 6c 3a 20 6e 6f 2d 63 61 63 68 65 0d 0a   trol: no-cache..
0090   0d 0a 00                                          ...
```
What i found is the very last `0x00` which does not exist in PC request, and it seems to send for some reason additional request (2 requests in total from 1). That second request is also getting response, but its always response 400.

[wireshark.zip](https://github.com/espressif/esp-idf/files/3444643/wireshark.zip)

chegewara, 2019-07-30:
Here is new wireshark file, i fixed my bug with trailing 0x0 in address:
[wireshark-esp32.pcapng.txt](https://github.com/espressif/esp-idf/files/3449064/wireshark-esp32.pcapng.txt)

playGit888, 2019-08-08:
I have the same problem, and posted a post before I saw this.
https://github.com/espressif/esp-idf/issues/3894
Anyone know how to fix it?

playGit888, 2019-08-08:
hi, farizadp, do you still have this problem?  wifi: ampdu: ignore deleting tx BA0
How you fix it?

farizadp, 2019-08-08:
> 
> 
> hi, farizadp, do you still have this problem? wifi: ampdu: ignore deleting tx BA0
> How you fix it?

@playGit888 : I still see that message. But we are currently not hampered by this.

zhangyanjiaoesp, 2019-08-08:
> I also have wireshark file from esp32 and similar request from PC browser.
> 
> EDIT that additional 0x00 was my mistake.
> 
> [wireshark.zip](https://github.com/espressif/esp-idf/files/3444643/wireshark.zip)

@chegewara We need the wifi captures for the `wifi: ampdu: ignore deleting tx BA0`, But the capture you provided is only for TCP transport.

zhangyanjiaoesp, 2019-08-08:
@farizadp 

> However, according to the discussion (see https://community.cisco.com/t5/other-wireless-mobility-subjects/deauth-reason-code-108/td-p/3374721).

 I can't open your link. 

>  it seems indeed that the router replied with 108. This is due to the device does not try to renew its DHCP lease.
Then we need to check why the device does not renew DHCP release. Maybe you should provide logs and capture packets.

farizadp, 2019-08-08:
> 
> 
> @farizadp
> 
> > However, according to the discussion (see https://community.cisco.com/t5/other-wireless-mobility-subjects/deauth-reason-code-108/td-p/3374721).
> 
> I can't open your link.
> 

Here is the snapshot
[image]

zhangyanjiaoesp, 2019-08-08:
So your problem is about roaming?

farizadp, 2019-08-08:
What does 'roaming to a new AP' means then?

zhangyanjiaoesp, 2019-08-08:
I saw it in your snapshot

farizadp, 2019-08-08:
Sorry, I was asking whether you have any idea what 'roaming to a new AP' means. And what should the ESP32 device do when it receives error code 108?

sagb2015, 2019-08-13:
"Roaming to new AP" means moving from one AP to another in the same ESS ( preferably with better signal strength)  without disrupting the connection.  You might be observing the issue only when there are multiple AP around with same SSID.

zhangyanjiaoesp, 2019-11-06:
@farizadp Do you still have the `error code 108` issue?  Are there multiple APs around with the same SSID when this issue  occurred?

Alvin1Zhang, 2020-05-29:
@farizadp Thanks for reporting. Feel free to reopen if the issue still happens or with more details. Thanks.

closed 2020-05-29 by Alvin1Zhang