espressif/esp-idf #3667: esp_ota_get_running_partition fail on OTA slot

anthodury, 2019-06-20:
## Environment

-Development Kit: esp32 devkit v1
-Module or chip used: ESP32-WROOM-32
-IDF version: v3.2-130-g8e51f7e23
-Build System: Make
-Compiler version: a1.22.0-80-g6c4433a
-Operating System: Linux VM on Windows host
-Power Supply: USB

## Problem Description

When I call the function "esp_ota_get_running_partition" when running on an OTA slot (ota_0) I get an assertion failure and the program abort.
When I call the same function on the factory partition I don't have this problem.

It seems that the "esp_partition_find" function return an empty result while executing on an OTA slot.

## Debug Logs

```
assertion "it != NULL" failed: file "/home/anthony/esp/esp-idf-v3-2/components/app_update/esp_ota_ops.c", line 515, function: esp_ota_get_running_partition
abort() was called at PC 0x4010513f on core 1
0x4010513f: _reclaim_reent at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/reent/../../../.././newlib/libc/reent/reent.c:89

Backtrace: 0x4008ed98:0x3ffb1e40 0x4008efc5:0x3ffb1e60 0x4010513f:0x3ffb1e80 0x401203a7:0x3ffb1eb0 0x40141894:0x3ffb1ed0 0x4013fb06:0x3ffb1f20 0x401204a7:0x3ffb1fb0 0x4008b4e5:0x3ffb1fd0
0x4008ed98: commonErrorHandler_dump at /home/anthony/esp/esp-idf-v3-2/components/esp32/panic.c:707

0x4008efc5: xPortGetCoreID at /home/anthony/esp/esp-idf-v3-2/components/esp32/panic.c:707
 (inlined by) panicHandler at /home/anthony/esp/esp-idf-v3-2/components/esp32/panic.c:221                                                                                                                                                   

0x4010513f: _reclaim_reent at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/reent/../../../.././newlib/libc/reent/reent.c:89

0x401203a7: uartEnd at /home/anthony/esp/factoryApp/components/arduino/cores/esp32/esp32-hal-uart.c:404

0x40141894: Best_Wifi::~Best_Wifi() at /home/anthony/esp/factoryApp/main/IOTLab_Wifi.h:23
 (inlined by) volvo::OTAESP32::getThingnameFromThingworxManager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/anthony/esp/factoryApp/main/OTA.cpp:86                                                                                                 

0x4013fb06: StringSumHelper::StringSumHelper(char const*) at /home/anthony/esp/factoryApp/main/IOTLab_Wifi.cpp:420
 (inlined by) connectToWifi(Best_Wifi, String, String, String) at /home/anthony/esp/factoryApp/main/IOTLab_Wifi.cpp:127                                                                                                                     

0x401204a7: uartWrite at /home/anthony/esp/factoryApp/components/arduino/cores/esp32/esp32-hal-uart.c:404

0x4008b4e5: vPortTaskWrapper at /home/anthony/esp/esp-idf-v3-2/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

chegewara, 2019-06-20:
> Return
Pointer to info for partition structure, or NULL if no partition is found or flash read operation failed. Returned pointer is valid for the lifetime of the application.

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html#_CPPv429esp_ota_get_running_partitionv

anthodury, 2019-06-20:
> https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html#_CPPv429esp_ota_get_running_partitionv

Thank you but I already know that, I read the documentation before creating the issue. 
According to the logs, the abort() on `"assertion "it != NULL"` is caused
because `"esp_partition_find(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_ANY, NULL);"` returned no partition. Why no partition is found? I don't access to the esp_partition_find code so I can search in this direction... I use the same code when on factory partition and it works but no when on a OTA partition.

chegewara, 2019-06-20:
What i wanted to show is that it can be NULL when `flash read operation failed`. 
Now did you encountered this issue 1 time or is it always with the same code?

anthodury, 2019-06-20:
> 
> 
> What i wanted to show is that it can be NULL when `flash read operation failed`.

Do you know how can I check that? I also have nvs init failure error that occur, maybe it's linked?

> Now did you encountered this issue 1 time or is it always with the same code?

I encounter this issue all the time if the program is currently running on the OTA partition. I call this function in order to then call  `esp_ota_get_partition_description` to get the version of the current running partition. When the ESP is currently running the factory partition, the `esp_partition_find` and then the `esp_ota_get_partition_description` calls both succeed and I can get the version of the application but not when on OTA partition.

chegewara, 2019-06-20:
Yes, this may be the clue. There is another thing, `esp_ota_get_partition_description` is not part of the esp-idf 3.2. Which you are using according to OP.

anthodury, 2019-06-21:
> Yes, this may be the clue. There is another thing, `esp_ota_get_partition_description` is not part of the esp-idf 3.2. Which you are using according to OP.

You are right, `esp_ota_get_partition_description` is not part of the esp-idf 3.2 that i use, I implemented it with the help of this commit https://github.com/espressif/esp-idf/commit/3b9cb25fe18c5a6ed64ddd6a1dc4d0ce0b6cdc2a#diff-36bac87c560c129a1b479e18c2cf14ba.

Unfortunately the issue is not caused by this modification.

I tried a simple thing: as a factory application I use the native_ota_example. For the OTA application I chose the hello_world example in which I added `nvs_flash_init()` and `esp_ota_get_running_partition()` calls as following:

```
// hello_world_main.c
void app_main()
{
    printf("Hello world!\n");
    
    // Initialize NVS.
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // OTA app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    
    ESP_LOGE("", "nvs_flash_init return err = %d", err);   
        
    const esp_partition_t * partition=esp_ota_get_running_partition();
    if(partition==NULL)
    {
        ESP_LOGE("", "esp_ota_get_running_partition returned NULL");
    }
    else
    {
        ESP_LOGI("", "Ok, esp_ota_get_running_partition succeeded");
    }
...

```

And after some experiments It's seems that the problem is caused by the partition table configuration.

Using this custom partition table .csv file:
```
# Espressif ESP32 Partition Table
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      0xc000,  0x4000
otadata,  data, ota,      0x10000,  0x2000
phy_init, data, phy,      0x12000,  0x1000
factory,  app,  factory,  0x20000,  1M
ota_0,    app,  ota_0,    ,         1M
nvs_key,  data, nvs_keys, ,        0x1000
```

If I set a partition table offset of 0x8000 in menuconfig like it is set by default it works, no errors in factory app or hello_world based OTA app.

But as soon as I increase the partition table offset for example from 0x8000 to 0x9000 the nvs_flash_init() fail (Return ESP_ERR_NOT_FOUND) and esp_ota_get_running_partition() fail when on OTA app (Still no problem when on factory app)

I want to use secure boot and flash encryption so I need to increase this offset because the bootloader become too big.

By default the offset is 0x8000 and the first partition is at 0x9000. 
By setting the offset to 0x9000 and the first partition at 0xA000 (nvs) or more it should work right?

KonstantinKondrashov, 2019-06-21:
Hi @anthodury! I can suggest you blank the offsets in you partition table [for more information](https://github.com/espressif/esp-idf/blob/master/components/partition_table/Kconfig.projbuild#L48-L500)

> To have each partition offset adapt to the configured partition table offset, leave all
partition offsets blank in the CSV file.

```
# Espressif ESP32 Partition Table
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      ,  0x4000
otadata,  data, ota,      ,  0x2000
phy_init, data, phy,      ,  0x1000
factory,  app,  factory,  ,  1M
ota_0,    app,  ota_0,    ,  1M
nvs_key,  data, nvs_keys, ,  0x1000
```

anthodury, 2019-06-21:
> Hi @anthodury! I can suggest you blank the offsets in you partition table [for more [...]

Hi @KonstantinKondrashov! 

I tried what you suggest but it doesn't solve the issue...
By using your proposal and an partition table offset of 0x8000 it's working but with an offset of 0x9000 it doesn't, same as above.

However, by doing a make `partition_table` with an offset of 0x9000 and the CSV you provided the output calculation seem good with a good:

```
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0xa000,16K,
otadata,data,ota,0xe000,8K,
phy_init,data,phy,0x10000,4K,
factory,app,factory,0x20000,1M,
ota_0,app,ota_0,0x120000,1M,
nvs_key,data,nvs_keys,0x220000,4K,
```

KonstantinKondrashov, 2019-06-21:
Yes, this part looks correct. Are you sure that your firmwares (1.factory_firmare and 2.hello_world_firmare) is compiled with the same partition table and offsets? Looks they have diffrent.

anthodury, 2019-06-24:
> Yes, this part looks correct. Are you sure that your firmwares (1.factory_firmware and 2.hello_world_firmware) is compiled with the same partition table and offsets? Looks they have diffrent.

My bad... I thought this part was only needed for the factory firmware as it is built and flashed at the same time as the partition table and the bootloader but I was wrong, the OTA app seems to also need that. By setting the same offset and partition table in both factory and OTA applications that's working! 

Thanks a lot!

However, just for my understanding, do you think it is really required to use the same partition table for the factory app and OTA app? I mean, in both cases I need to use a custom .CSV file overwise I can't compile due to overlaps error but appart from that, is the partition table important when building the OTA application as it will not be flashed?

KonstantinKondrashov, 2019-06-24:
When you use OTA, the partition table and the offset of table have to be the same for the bootloader, the factory_app, and ota_app. The partition table is flashed only first time and will not be updated anymore in the future. Any app/factory/bootloader have to know the offset of table.

anthodury, 2019-06-25:
Thank you for the precision.

closed 2019-06-25 by anthodury

SalmanMalik7183, 2023-01-26:
> When you use OTA, the partition table and the offset of table have to be the same for the bootloader, the factory_app, and ota_app. The partition table is flashed only first time and will not be updated anymore in the future. Any app/factory/bootloader have to know the offset of table.

Hi  @KonstantinKondrashov @chegewara @anthodury , 

It's been a while since you guys closed this issue, but i'm facing a similar problem with my code. 
I am trying to flash a signed firmware image into the esp32-c3 Mini-1 board.  

The firmware image in created through Arduino framework, and than signed with a RSA secure boot key and than flashed on to to board using idftool.py utility.
I am getting error here with the partition table it says the following.

[image]

SalmanMalik7183, 2023-01-26:
I am using the following .csv file for the partition table...

`# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap,,,,
nvs,        data,     nvs,      0xb000,        0x5000,
otadata,    data,     ota,      0x10000,        0x2000,
phy_init,   data,     phy,      0x12000,        0x1000,
factory,    app,      factory,  0x20000,       0x100000,
ota_0,      app,      ota_0,    0x120000,      0x100000,
ota_1,      app,      ota_1,    0x220000,      0x100000,
`

SalmanMalik7183, 2023-01-26:
The partition table is uploaded into the bootloader just fine the problem occurs in the following steps:

[image]

KonstantinKondrashov, 2023-01-26:
Hi @SalmanMalik7183!
It is not a similar issue. Please create a new issue.
Have you used ``--disable-md5sum`` option for ``gen_esp32part.py``?

SalmanMalik7183, 2023-01-26:
> Have you used `--disable-md5sum` option for `gen_esp32part.py`?

Yes I have disabled the md5 in the menuconfig.