espressif/esp-idf #3669: Regression: spi_flash Kconfig broken in v3.3-beta3

jmattsson, 2019-06-21:
## Environment

- Development Kit:   none
- IDF version (run ``git describe --tags`` to find it):   v3.3-beta1-694-g6b3da6b18 (v3.3-beta3 tag used to checkout)
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a5
- Operating System:     Linux

## Problem Description

Commit 37126d3451eabb44eeeb48b8e2ee554dc233e2a8 by @dobairoland changed the indentation in the Kconfig file, and broke the `SPI_FLASH_WRITING_DANGEROUS_REGIONS` options. The config options are now treated as part of the help text. This is Not Good(tm).

I have not checked whether this commit also broke other areas. It seems quite possible though.

### Expected Behavior

Being able to configure the `SPI_FLASH_WRITING_DANGEROUS_REGIONS` options.

### Actual Behavior

The option does not show any alternatives.

### Steps to repropduce

`make menuconfig`
-> components -> SPI Flash driver -> Writing to dangerous flash regions

negativekelvin, 2019-06-21:
https://github.com/espressif/esp-idf/issues/3598

jmattsson, 2019-06-21:
Doh, I only searched for open issues. Sorry about the dupe.

projectgus, 2019-06-21:
Hi @jmattsson ,

Sorry, we had a hold-up in our deployment system which meant the fix @dobairoland made in 60b937b5caa3b6d6376d91bf3ad8a508d06dbd7d has only just pushed to GitHub.

However it should be fixed in the release/v3.3 branch now.

Thanks,

Angus

closed 2019-06-21 by projectgus

jmattsson, 2019-06-21:
Great, thanks Angus!